arduino/Arduino #4520: EEPROM function definitions

cvmanjoo, 2016-02-03:
Where can i find function definitions for EEPROM Libraries.
I'm able to find EEPROM.h file but not the .c file. Can one help.?

Thanks,
Sorry If i posted this question is wrong place.

agdl, 2016-02-03:
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries/EEPROM

Here. I'm closing this

closed 2016-02-03 by agdl

Chris--A, 2016-02-03:
There is no .c or .cpp file, it is entirely implemented in the header!

cvmanjoo, 2016-02-03:
@Chris--A  Ya got it thanks.

q2dg, 2016-02-03:
I don't understand the meaning of https://github.com/arduino-libraries, then... EEPROM library isn't there although other seems to be...

Chris--A, 2016-02-04:
That is for exclusively owned Arduino LLC libraries. You'll notice all the repos have *"Copyright (c) Arduino LLC. All right reserved."*.

Whereas that claim cannot be placed on this library at it has been written by myself and attributed to the fact in the licence. The library is just being used by the Arduino AVR core.

q2dg, 2016-02-04:
Oh, I see. 
Thanks a lot for the clarification!