arduino/Arduino #4523: Atmel ICE Not Recognized

Pecacheu, 2016-02-04:
I recently picked up an Atmel ICE, and I'm trying use it to burn the Arduino bootloader to a Zero-like board with a ARM chip on it. I've tried both the `Atmel-ICE` and `Atmel-SAM-ICE` settings and they both fail with the generic "**Error while burning bootloader.**" The console doesn't show anything useful either. Worse yet, I don't even see the ICE show up in the ports menu when I plug it in. It also isn't powering the target, but that might be normal. The power LED does come on when I plug it in.

aethaniel, 2016-04-29:
@Pecacheu, you may have some lines in the console describing the error. Can you report them here, please?
Also, is the ARM chip on your custom board a SAMD21G18 or another product?

facchinm, 2016-07-22:
@Pecacheu , does it still apply to latest SAMD core? If so, I'll move the issue to the correct repo (https://github.com/arduino/ArduinoCore-samd)

Pecacheu, 2016-07-23:
No. This issue was fixed a while ago actually. There is a problem when using the ICE where Arduino occasionally says the chip could not be reset properly or something (I don't have my ICE with me at the moment to check exactly what it was), but the program uploads successfully anyway.

The only other issue I have with the SAMD core is this odd warning message that gets spewed out no matter what version I use: `Warning: platform.txt from core 'Arduino SAMD (32-bits ARM Cortex-M0+) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.`

closed 2016-07-23 by Pecacheu