arduino/Arduino #4526: Arduino Zero - 1 i2c interface used out of the box

dhiltonp, 2016-02-05:
It turns out that the Zero's EDBG chip is connected as a slave to the MCU, and shares the same address as a device I was trying to use (0x28).

-----------------------------------

I consider this a design flaw. The EDBG chip's i2c interface is intended to be used by the PC debugger, and I've been unable to find any documentation on what the chip can do and how to talk to it.

If there *is* a use case, it should be documented. If there is no use case, the lines should be removed and it should be documented. If there is a rare use case, maybe solder jumpers should be added and it should be documented.

Did I mention that it should be documented?

-----------------------------

Thanks!

Fede85, 2016-02-05:
Hi David,

Thanks for your reporting this.
On the Zero the DGI interface of the EDBG chip is connected to some of the peripherals of the SAMD21.
These lines were added to provide all the debugging features offered by Atmel Studio. At the moment there's any real use case in the Arduino enviroment, so thank to the feedback from the community like yours, we are evaluating to remove it in a future release.

For the moment I mark this issue as a documentation improvement.

Best regards,
Federico

closed 2016-02-05 by Fede85

carlosperate, 2016-02-23:
> At the moment there's any real use case in the Arduino enviroment, so thank to the feedback from the community like yours, we are evaluating to remove it in a future release

Okay, you can't tell me you are surprised about users not taking advantage of a new hardware feature with __no means to use it with the official Arduino Software__, and not having it __properly documented other than "it's there"__.

The debugger is most definitely a step in the right direction, debugging Arduino software is one of the complains most users highlight when discussing the advatages/disadvantages of Arduino, and I'm assuming I wouldn't be the only one sad to see it go due to lack of support.

dhiltonp, 2016-02-23:
This isn't about the debug chip itself, it's about having the debug chip attached to the MCU's i2c. The debug chip's i2c is for communication to the debugger, not to monitor i2c traffic (according to the documentation I've seen).

erichbieri, 2017-09-02:
I just stumbled over the same problem using a MAXIM digital potentiometer e.g. MAX5477. 
They use 0x28 as the default address too. Fortunately you can change the I2C address. 
Setting the new pot values randomly failed. This should defenatly be documented!