arduino/Arduino #4527: Request for Fixed Y scale on serial plotter

Bort-Plate, 2016-02-05:
With values that constantly change, the range of the plot will shrink and expand, making it incredibly difficult to judge the amplitude of a plot without constantly re-checking the Y-axis (of which is too small to read on my 4K screen anyway).

By having a selectable minimum and maximum, each peak on the plot can be visually compared properly.

cousteaulecommandant, 2016-02-10:
I agree that this is annoying; if not a selectable range (which I think would be best), at least an option to disable auto-scaling.

Also a horizontal grid would help.

karlTH, 2016-02-11:
serial plotter is an awesone tools
a csv export will be really good.
thanks a lot for the work.
i m also agree with cousteaulecommandant

andreashove, 2016-12-07:
A quick fix to this is to print your desired scale every time you print a value. 

If your values are between 0 and 1000 you can do:

```
String withScale = "0 ";
withScale += value;
withScale += " 1000";

Serial.println(withScale);
```

MickMad, 2017-02-20:
I totally agree. Autorange is no good for me. I am currently sending a value between 0.0 and 1.0 to the serial plotter and it autoranges to -6.0 to +6.0 ... Why?

@andreashove we should not include a lot of overhead in memory and time of execution to our codes because a basic feature is missing.

phoenixperry, 2017-03-16:
I am going to agree that this is really difficult to work with and could use a fix.

alecokas, 2017-10-15:
I agree with @MickMad. Has any progress been made on this issue? I think it would be a useful addition.

steeley, 2019-03-16:
yep, 3 years on and still no option....
Plotter is very useful, but on/off autoscale should have been a fundamental feature.
I had to hack  together my own plotter as part of a desktop app, but would be easier
in Arduino IDE.