arduino/Arduino #4529: IDE re-attaches to non-existent COM port after upload on M0 board

cuyahoga, 2016-02-05:
After compiling and uploading any sketch - Blink will do as an example - the selected port jumps from the correct one (say COM13) to the next numerical - but non-existent - port, (such as COM14). The sketch gets uploaded, but one then has to wait for the board to finish re-initialising after the upload, before re-selecting the correct port in the IDE. Same thing happens if the board's on COM17 ... it jumps to COM18.

OS : Win 7 64-bit
IDE : 1.6.7 and nightly (2016/02/02)
SAMD Boards 1.6.2: 

Verbose output on upload revealed;

> Sketch uses 10,988 bytes (4%) of program storage space. Maximum is 262,144 bytes.
> Forcing reset using 1200bps open/close on port COM13
> PORTS {COM3, COM13, } / {COM3, COM13, } => {}
> PORTS {COM3, COM13, } / {COM3, } => {}
> PORTS {COM3, } / {COM3, } => {}
> **PORTS {COM3, } / {COM3, COM14, } => {COM14, }
> Found upload port: COM14
> C:\Users\xxxxx\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe -i -d --port=COM14 -U true -i -e -w -v C:\Users\xxxxx~1\AppData\Local\Temp\build81716e661910f192b156ff7c623177af.tmp/Blink.ino.bin -R**
> Set binary mode
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> version()=v1.1 [Arduino:XYZ] Oct 31 2015 00:47:48
> chipId=0x10010005
> Connected at 921600 baud
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> Atmel SMART device 0x10010005 found
> write(addr=0x20004000,size=0x34)
> writeWord(addr=0x20004030,value=0x10)
> writeWord(addr=0x20004020,value=0x20008000)
> Device       : ATSAMD21G18A
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> Chip ID      : 10010005
> version()=v1.1 [Arduino:XYZ] Oct 31 2015 00:47:48
> Version      : v1.1 [Arduino:XYZ] Oct 31 2015 00:47:48
> Address      : 8192
> Pages        : 3968
> Page Size    : 64 bytes
> Total Size   : 248KB
> Planes       : 1
> Lock Regions : 16
> Locked       : readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> none
> readWord(addr=0x41004018)=0
> Security     : false
> Boot Flash   : true
> readWord(addr=0x40000834)=0x7000a
> BOD          : true
> readWord(addr=0x40000834)=0x7000a
> BOR          : true
> Arduino      : FAST_CHIP_ERASE
> Arduino      : FAST_MULTI_PAGE_WRITE
> Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
> Erase flash
> chipErase(addr=0x2000)
> done in 0.839 seconds
> 
> Write 11136 bytes to flash (174 pages)
> write(addr=0x20005000,size=0x1000)
> writeBuffer(scr_addr=0x20005000, dst_addr=0x2000, size=0x1000)
> 
> [===========                   ] 36% (64/174 pages)write(addr=0x20005000,size=0x1000)
> writeBuffer(scr_addr=0x20005000, dst_addr=0x3000, size=0x1000)
> 
> [======================        ] 73% (128/174 pages)write(addr=0x20005000,size=0xb80)
> writeBuffer(scr_addr=0x20005000, dst_addr=0x4000, size=0xb80)
> 
> SAM-BA operation failed
> 
> [==============================] 100% (174/174 pages)
> done in 0.099 seconds
> 
> Verify 11136 bytes of flash with checksum.
> checksumBuffer(start_addr=0x2000, size=0x1000) = 8257
> checksumBuffer(start_addr=0x3000, size=0x1000) = 66c7
> checksumBuffer(start_addr=0x4000, size=0xb80) = a5f8
> Verify successful
> done in 0.051 seconds
> CPU reset.
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> writeWord(addr=0xe000ed0c,value=0x5fa0004)
>

facchinm, 2016-02-08:
Hi @cuyahoga ,
this issue is a duplicate of #4367 and probably of another couple of issues I'm unable to find right now. It should be solved by #4482, if you could test [this](https://github.com/arduino/Arduino/pull/4482#issuecomment-180330393) IDE version and report the result here it would be great!

cuyahoga, 2016-02-08:
My bad, thought I'd checked for a pre-existing issue, sorry.

Afraid that branch doesn't fix the problem. The board was on COM17, uploaded Blink and the port jumped to COM18 still. Deleted the cache in AppData before trying, in case there was anything in there affecting it. Anything else I should have done?

> Sketch uses 10,988 bytes (4%) of program storage space. Maximum is 262,144 bytes.
> Forcing reset using 1200bps open/close on port COM17
> PORTS {COM3, COM17, } / {COM3, } => {}
> PORTS {COM3, } / {COM3, } => {}
> PORTS {COM3, } / {COM3, } => {}
> PORTS {COM3, } / {COM3, COM18, } => {COM18, }
> Found upload port: COM18
> C:\Users\xxxxx\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe -i -d --port=COM18 -U true -i -e -w -v C:\Users\xxxxx\AppData\Local\Temp\buildc47db500b84fc8cde939aff867398723.tmp/Blink.ino.bin -R 
> Set binary mode
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> version()=v1.1 [Arduino:XYZ] Oct 31 2015 00:47:48
> chipId=0x10010005
> Connected at 921600 baud
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> Atmel SMART device 0x10010005 found
> write(addr=0x20004000,size=0x34)
> writeWord(addr=0x20004030,value=0x10)
> writeWord(addr=0x20004020,value=0x20008000)
> Device       : ATSAMD21G18A
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> Chip ID      : 10010005
> version()=v1.1 [Arduino:XYZ] Oct 31 2015 00:47:48
> Version      : v1.1 [Arduino:XYZ] Oct 31 2015 00:47:48
> Address      : 8192
> Pages        : 3968
> Page Size    : 64 bytes
> Total Size   : 248KB
> Planes       : 1
> Lock Regions : 16
> Locked       : readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> readWord(addr=0x41004020)=0xffff
> none
> readWord(addr=0x41004018)=0
> Security     : false
> Boot Flash   : true
> readWord(addr=0x40000834)=0x7000a
> BOD          : true
> readWord(addr=0x40000834)=0x7000a
> BOR          : true
> Arduino      : FAST_CHIP_ERASE
> Arduino      : FAST_MULTI_PAGE_WRITE
> Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
> Erase flash
> chipErase(addr=0x2000)
> done in 0.839 seconds
> 
> Write 11136 bytes to flash (174 pages)
> write(addr=0x20005000,size=0x1000)
> writeBuffer(scr_addr=0x20005000, dst_addr=0x2000, size=0x1000)
> 
> [===========                   ] 36% (64/174 pages)write(addr=0x20005000,size=0x1000)
> writeBuffer(scr_addr=0x20005000, dst_addr=0x3000, size=0x1000)
> 
> [======================        ] 73% (128/174 pages)write(addr=0x20005000,size=0xb80)
> writeBuffer(scr_addr=0x20005000, dst_addr=0x4000, size=0xb80)
> 
> SAM-BA operation failed
> 
> [==============================] 100% (174/174 pages)
> done in 0.120 seconds
> 
> Verify 11136 bytes of flash with checksum.
> checksumBuffer(start_addr=0x2000, size=0x1000) = 8257
> checksumBuffer(start_addr=0x3000, size=0x1000) = 66c7
> checksumBuffer(start_addr=0x4000, size=0xb80) = a5f8
> Verify successful
> done in 0.051 seconds
> CPU reset.
> readWord(addr=0)=0x200023a8
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> writeWord(addr=0xe000ed0c,value=0x5fa0004)
> Board at COM18 is not available

facchinm, 2016-04-11:
Hopefully solved by merging #4792

closed 2016-04-11 by facchinm