arduino/Arduino #4533: Keyboard.h not working and cannot be included.

mikila85, 2016-02-06:
on 1.6.5
trying to use Keyboad Lib and getting this error:

C:/Program Files/Arduino/libraries/Keyboard/src/Keyboard.h:29:2: warning: #warning "Using legacy HID c

NicoHood, 2016-02-06:
Please use the latest Arduino IDE and try again.

facchinm, 2016-02-12:
Hi @mikila85 ,
the Keyboard library comes with IDE 1.6.6, please update to latest IDE as suggested by @NicoHood to get rid of the warning (and to get Keyboard functionality)

closed 2016-02-12 by facchinm

akshaygugnani15, 2020-07-07:
You know what you don't require the 32u4 now, I got a solution to use the keyboard and mouse through any Arduino yes you read it right any!

### All you have to do is something like this on the Arduino:-

`Serial.println("pressA")`

### And now head to a python script and do this:-

```
import serial
import pyautogui
Arduino_Serial = serial.Serial('COM5', 9600)

while 1:
    incoming_data = str(Arduino_Serial.readline())
    print(incoming_data)
    if 'pressA' in incoming_data:
        pyautogui.press('a')
    incoming_data = ""
```

BOOM YOU ARE DONE