arduino/Arduino #4534: Conflict Between Wifi101.h and SoftwareSerial.h

mjmoore10, 2016-02-06:
If I try to use wifi101.h and softwareserial.h in the same sketch, I get the error messages below from the IDE. I'm a beginner, so I may well be wrong, but it looks to me like both header files define either functions or variables with the same names, which causes the conflict. If there's really a bug, I hope this report is helpful. If I'm just ignorant and this is an avoidable problem, I'd be grateful for any guidance.

Arduino: 1.6.7 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

libraries\SoftwareSerial\SoftwareSerial.cpp.o: In function `__vector_9':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:229: multiple definition of `__vector_9'

libraries\WiFi101\bsp\source\nm_bsp_arduino_avr.c.o:<path info>/nm_bsp_arduino_avr.c:67: first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

libraries\SoftwareSerial\SoftwareSerial.cpp.o: In function `SoftwareSerial::read()':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:394: multiple definition of `__vector_10'

libraries\WiFi101\bsp\source\nm_bsp_arduino_avr.c.o:<path info>/nm_bsp_arduino_avr.c:62: first defined here

libraries\SoftwareSerial\SoftwareSerial.cpp.o: In function `SoftwareSerial::read()':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:394: multiple definition of `__vector_11'

libraries\WiFi101\bsp\source\nm_bsp_arduino_avr.c.o:<path info>/nm_bsp_arduino_avr.c:62: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling.

facchinm, 2016-02-08:
Hi @mjmoore10,
thanks for spotting this bug! 
In the meantime, if you only need to use a Mega, you can open ```libraries\WiFi101\bsp\source\nm_bsp_arduino_avr``` with a text editor end remove lines 
```
#if defined(PCINT0_vect)
ISR(PCINT0_vect)
{
	if (!rx_pin_read() && gpfIsr)
	{
		gpfIsr();
	}
}
#endif

#if defined(PCINT1_vect)
ISR(PCINT1_vect, ISR_ALIASOF(PCINT0_vect));
#endif

#if defined(PCINT2_vect)
ISR(PCINT2_vect, ISR_ALIASOF(PCINT0_vect));
#endif

#if defined(PCINT3_vect)
ISR(PCINT3_vect, ISR_ALIASOF(PCINT0_vect));
#endif
```

This code is UNO specific and shouldn't be compiled for Mega.

mjmoore10, 2016-02-08:
I'm delighted to have helped in some small way. And thank you very much for
the workaround!

closed 2016-02-19 by agdl