arduino/Arduino #4537: SM Library

MichelRichir, 2016-02-07:
SM library works fine with Arduino 1.6.5 but not with Arduino 1.6.7

per1234, 2016-02-07:
Since you haven't provided any information of where you downloaded the SM library(http://playground.arduino.cc/Code/SMlib ?) or what the problem you're encountering is I have to do some guesswork but I believe this  is a duplicate of https://github.com/arduino/arduino-builder/issues/50. So, using the example_2 example, you would need to change the line:
```arduino
SM Simple(S1);//create simple statemachine
```
to:
```arduino
SM Simple(&S1);//create simple statemachine
```

agdl, 2016-02-08:
[Looking at the history](http://playground.arduino.cc/Code/SMlib?action=diff) maybe you can ask to the contributor to make the change or maybe someone can upload the library on Git and insert it in the library manager in this way you don't have to add it manually and someone else can easily find it

per1234, 2016-02-08:
I agree it would be good to update the library. There is also an issue with example_1.ino where the case of the folder name doesn't match the case of the filename so it won't compile. @agdl do you know whether it is possible to do attachments on the Playground anymore? I just did a test in the sandbox http://playground.arduino.cc/Main/WikiSandbox but when I click on that link to do the upload the password prompt won't let me past with the prefilled password. I'll send the library author a PM to ask if they'd like to update the library but I'd like to be able to give them the correct information regarding the attachment issue. I always prefer to see libraries on GitHub and I am willing to create a repository for the library if the author has abandoned it.

agdl, 2016-02-08:
@per1234 I'm too too expert of playground, but maybe if there are issue @mastrolinux can help!

mastrolinux, 2016-02-15:
Hi @per1234 the upload feature is restricted to some users only. If they can provide us an updated file we can upload it.
Thank you.

per1234, 2016-02-15:
@mastrolinux thanks for the information. I think that was a good idea. It's better for files to be hosted elsewhere and just put a link to it on the playground. I have documented that regular users are no longer allowed to attach files in the Participate page of the Playground.

I sent the library author a PM via the Arduino website to notify them of this issue.

agdl, 2016-02-22:
@per1234 can this be closed?

per1234, 2016-02-22:
@agdl I think so. I haven't received a response to my message to the author but another user has created a repository for the library https://github.com/dangu/SMlib and we have fixed the examples there so they will compile with Arduino IDE 1.6.6+.

closed 2016-02-22 by agdl

dangu, 2016-02-23:
@per1234: May I kindly ask you to remove your SMlib-repo from github. I apparently did something horribly wrong creating this repository.

per1234, 2016-02-23:
@dangu I have done so. The purpose of my fork was only to contribute the pull requests I made to your repository and I had just not gotten around to deleting it yet.

So just to make it clear to others who may encounter this issue report in the future, the SM library has been removed by the author from the Arduino Playground as well as all copies of it from GitHub and is no longer publicly available. So this means this issue has been resolved in a completely different manner than I had stated in my last comment.