arduino/Arduino #4538: 1.6.8  Build 2016/02/02 09:41

VaclavSal, 2016-02-07:
Could someone please give me some clue why I cannot compile on 1.6.8 ?
I have been using 1.6.6 and feel it is time to experiment with something more recent.  

Did I missed something? 

Arduino: 1.6.8 Hourly Build 2016/02/02 09:41 (Windows XP), Board: "Arduino Due (Programming Port)"

F:\Arduino167\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware "F:\Arduino167\arduino-nightly\hardware" -hardware "C:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages" -tools "F:\Arduino167\arduino-nightly\tools-builder" -tools "F:\Arduino167\arduino-nightly\hardware\tools\avr" -tools "C:\Documents and Settings\Vaclav\Local Settings\Application Data\Arduino15\packages" -built-in-libraries "F:\Arduino167\arduino-nightly\libraries" -libraries "C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries" -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10608 -build-path "C:\DOCUME~1\Vaclav\LOCALS~1\Temp\build1e5cc712f8df906159907d9814e68fab.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "C:\Documents and Settings\Vaclav\My Documents\Arduino\SerialEvent_2\SerialEvent_2.ino"
Exception 0xc000001d 0x1003f 0x0 0x4d46d9
PC=0x4d46d9

math.initÂ·1()
	c:/go/src/math/pow10.go:34 +0x19
math.init()
	c:/go/src/math/unsafe.go:21 +0x3f
strconv.init()
	c:/go/src/strconv/quote.go:455 +0x45
io/ioutil.init()
	c:/go/src/io/ioutil/tempfile.go:95 +0x4a
github.com/go-errors/errors.init()
	c:/jenkins/workspace/arduino-builder-windows/src/github.com/go-errors/errors/stackframe.go:102 +0x3f
main.init()
	c:/jenkins/workspace/arduino-builder-windows/main.go:395 +0x3f

goroutine 2 [runnable]:
runtime.forcegchelper()
	c:/go/src/runtime/proc.go:90
runtime.goexit()
	c:/go/src/runtime/asm_386.s:2287 +0x1

goroutine 3 [runnable]:
runtime.bgsweep()
	c:/go/src/runtime/mgc0.go:82
runtime.goexit()
	c:/go/src/runtime/asm_386.s:2287 +0x1
eax     0x10d0c000
ebx     0x635e68
ecx     0x10d02000
edx     0x635e68
edi     0x0
esi     0x52b0
ebp     0x0
esp     0x10d0bf80
eip     0x4d46d9
eflags  0x10202
cs      0x1b
fs      0x3b
gs      0x0
arduino-builder returned 2

Error compiling.
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\Advanced: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\Advanced
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT_Due: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT_Due
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\Advanced: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\Advanced
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\FATS
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minmsc_2013_11_27_v1_1
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\TFT_V2
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\USB_MSC_Minimal
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT
Invalid library found in C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT_Due: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\UTFT_Due

facchinm, 2016-02-08:
Hi @VaclavSal ,
the issue is a duplicate of #4251 and it's due to running the IDE on a processor without SSE2 instructions (Pentium 3 or Athlon XP). We are trying to bypass this problem (which is a golang embedded feature)

closed 2016-02-08 by facchinm

VaclavSal, 2016-02-08:
Thanks for the update