arduino/Arduino #4539: Java 8 ??

VaclavSal, 2016-02-07:
I like to start doing IDE development. 
Noted that 1.6.6.  "switched  to " Java 8. 
I have XP and cannot even get SDK SE 7 to execute.
I have no warm feeling for XP  and do not want to use Windows 10, but I willing to switch to Linux.
Would that be better way to go than back-paddle to whatever Java will work on XP?

facchinm, 2016-02-08:
Hi @VaclavSal ,
the switch to Java8 allowed us to solve a lot of problems and to simplify the IDE in a lot of places by using java8 specific constructs, so there is no safe way to revert using an older java version.
If you are interested in developing the IDE itself, I strongly encourage you to switch to Linux :smile:

closed 2016-02-08 by facchinm

VaclavSal, 2016-02-08:
OK, what you are saying is that software development tool  , such as Java , is OS dependent.
It does not surprise me , however, will I have a similar problems with Linux? 
Or do you recommend Linux just because the "upgrades" are free? 
Thanks

facchinm, 2016-02-09:
No, I'm strongly supporting Linux for developing the IDE since compiling is as easy as typing 
``` cd build && ant build run``` and the prerequisite are something like ```apt-get install openjdk8 ant``` instead than messing up with hidden configurations and environment variables :smile: