arduino/Arduino #4540: Can't create/use a project with ANY name

Gargo, 2016-02-07:
It doesn't allow to use .ino file until it will be in the folder with the same name. BUT WHY?

agdl, 2016-02-09:
@Gargo because this is the arduino standard: you must have a folder that is named as the main .ino file. In the same folder you can also have other .ino files but also .h, .cpp .s etc.
The file with the same name as the folder is the main file. Actually this isn't an issue so I'm closing this!

closed 2016-02-09 by agdl

Gargo, 2016-02-09:
You have closed it too early.
Okay, I want to copy a folder which contains my project into the same folder. It becomes a hell if this folder contains a lot of files. Could you suggest an appropriate solution?

PaulStoffregen, 2016-02-09:
> becomes a hell if this folder contains a lot of files. Could you suggest an appropriate solution?

Maybe use Ctrl-A (or Cmd-A on a Macintosh) to select all the files.  Then press and hold your mouse button on any of the file, and use a movement of the mouse to drag the entire lot to the new location.  Is that so hard?

Gargo, 2016-02-09:
The same trouble - the parent folder will have the different name and I would have a possibility to place ino files into the same folder with other. Or do you suggest to have 2 folders inherited into each other? Maybe but there is no other ide which is so "user friendly"

matthijskooijman, 2016-02-17:
I'm not exactly sure I understand the question exactly, but you can open up a sketch and use "save as" to copy it to a different name, or use "rename" in the tab menu (the arrow to the right of the tab bar) to rename the sketch.

Gargo, 2016-02-18:
yes. And after these actions try to open it

matthijskooijman, 2016-02-18:
I'm pretty sure I did that recently, without any problems. Is save as / rename not working for you in the IDE?

Gargo, 2016-02-18:
The problem is I can't open the project file except of the situation when it is in a folder with the same name