arduino/Arduino #4545: Arduino Genuino 101 (Intel) doesn't work with Arduino WiFi Shield 101

Walter82, 2016-02-08:
Hello,
I attached an Arduino WiFi Shield 101 to my Genuino 101. It doesn't work. The same sketch works on Arduino UNO and on Arduino Zero. But it doesn't work on Genuino 101. There is nothing happening after the command WiFi.begin(ssid, pass). I can not run any WiFi Shield 101 example sketch at all. Please help. What could be the problem? Thank you very much.

facchinm, 2016-02-09:
Hi @Walter82 ,
I've just tested a WiFi101 with a Genuino 101 and it works correctly. Can you add some details on your configuration? If you don't mind, I'm moving this issue to the forum since it's not related with the IDE and some there is more chance to get support from other users there.
The imported topic is [here](http://forum.arduino.cc/index.php?topic=377942.0)

closed 2016-02-09 by facchinm

Walter82, 2016-02-09:
Hello, thank you very much.