arduino/Arduino #4546: Library update notifications on IDE 1.6.x

clovisf, 2016-02-08:
I personally did not like how **invasive** the "Library update" notifications popup (of the Arduino IDE) is. The suggestions is to pop a message during the IDE startup or even not to notify about that at all.

facchinm, 2016-02-09:
@00alis

lmihalkovic, 2016-02-09:
IMHO the problem is to to-notify-or-not-NOTIFY but more an issue of very poor execution. Not every app with pops notifications (even those written in Java with poor ol'swing) behave so intrusively. In many sutuations the feedback is cleanly hidden after a period (sometimes configurable) with the possibility if viewing past ones.

00alis, 2016-02-09:
The execution is indeed really poor. We will schedule e redesign of the behavior in the next few weeks, and see how easy will be to use native OS notification systems. Thanks!

per1234, 2016-02-10:
It would be nice to have an ignore option for the updatable library/board notifications. If clicked this would suppress all further notifications of that specific version.

leomil72, 2016-02-21:
I agree with clovisf, the pop-up, when it appears, stays over the rest of the windows' application, covering everything and making really impossible to interact with the GUI until you close it. (Mac OSX)

pfeerick, 2016-03-30:
Yes, indeed... the pop-up is very annoying in that it appears over the top of everything (on Windows 10). So if you opened the IDE, minimised it, and then was using your web browser, etc when the update check finished, it pops up at you and forces you to close it to get rid of it, rather than hiding nicely in the background over the Arduino IDE.

lmihalkovic, 2016-03-30:
The problem has been identified a few months ago and has a simple. I don't understand what's holding putting it to rest once and for all?

pfeerick, 2016-03-31:
More like 1 & 1/2 months ago... but I get the idea. It has been acknowledged, and the UI design folks are looking into using the official OS notifications instead (hence the comment from Alice on the 10th of Feb), which will make it much less annoying. So it shouldn't take too much longer! :-)

lmihalkovic, 2016-03-31:
@pfeerick I believe this issue was referenced in older entrie(s)

cmaglie, 2016-04-04:
I've pushed this fix (now available on the hourly build):
- The "invasive" popup is no more always-on-top https://github.com/arduino/Arduino/commit/1a7eaaf8b2208e04ec4d19c369f1ea277ac35f75
- Auto close after 10 seconds. https://github.com/arduino/Arduino/commit/8d7ee63c70e7b6d2cddaa741b05decfda0bd78ab

Still not the perfect solution but at least it should fixes all the annoying side effects.

pfeerick, 2016-04-05:
@cmaglie The changes you've made look good... but now I don't have any libraries or boards wanting updates, so can't see it in action! :-O As far as I'm concerned, this should make it near enough to perfect - it won't pop up over the top of everything, and if you ignore it for 10 seconds it will go away. basically the same behaviour of system notifications, whilst remaining platform agnostic.

sandeepmistry, 2016-07-18:
Closing this @cmaglie's changes have been merged into master and released.

closed 2016-07-18 by sandeepmistry