arduino/Arduino #4548: Add MQTT PubSub Client to in Library Manager

hirotakaster, 2016-02-09:
Hi, 
This MQTT library for Arduino is compatible with Spark Core and Photon(cross platform), over 1000 apps use this library now. It would be good this library can use in library list.
https://github.com/hirotakaster/MQTT/tree/arduino
Thanks!

agdl, 2016-02-09:
@hirotakaster first of all you have to create a tag (please follow this guide: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) and then please give the library a less generic name like hirotakasterMQTT

hirotakaster, 2016-02-09:
@agdl I fixed your check. I create the v1.3.5 tag, and library name change to MQTTCilent.

agdl, 2016-02-09:
I can't see the library name change, however MQTTClient is still a too generic name

hirotakaster, 2016-02-09:
Do you mean repository name change to specified name? not a library.properties file?

agdl, 2016-02-09:
Usually names are compliant between repository and library.properties... However you made a mistake in the name you wrote Cilent

If someone else add a library with the same name choosing one is a mess

agdl, 2016-02-09:
If you fix the tag name with Client instead of Cilent I will add it however since the contributor name is well visible

hirotakaster, 2016-02-09:
Okay, I know. I will upload different repository name in next time. Please close this Issue. Thank you.

closed 2016-02-09 by agdl