arduino/Arduino #4551: File edited externally to IDE does not reload

tjayrush, 2016-02-09:
I frequently edit my code in both the IDE and external to the IDE when I program in xCode (not necessary to ask why--trust me--it makes sense).

In xCode, this works fine because the file reloads (assuming it's been saved in the IDE--it warns me otherwise), if I edit it externally.

In Arduino IDE the file does not reload (either automatically or after a warning). In other words, the Arduino IDE is ignoring OS X's message that the file has changed.

Suggestion: IDE should watch for and react to externally edited files and update the file in the IDE if it changes. This is, what I would call, "standard operating procedure."

lmihalkovic, 2016-02-09:
Aside from the 'simple' possibility of a manual chk upon window activation, there is a standard java API for doing that cross platform.

tjayrush, 2016-02-09:
Sorry for the newbie question (new to GitHub and Arduino). Does your comment mean "Cool idea, go ahead and implement that in your [me] own version," or "Good idea. I [Arduino developers] will make that change"?

As I say, I'm new to this whole thing. Thanks.

chrstphrchvz, 2016-02-09:
(Another user here.) Note that in Preferences there is the "Use external editor" option that causes the editor to be read-only, and **does** refresh project to refresh upon focusing on the Arduino IDE--but this is not an option if you still need to use the Arduino IDE's editor. I agree that this is somewhat non-standard where other editors such as XCode, TextWrangler watch for file changes rather than application focus, and allowing out-of-the-box compatibility with git.

lmihalkovic, 2016-02-10:
```
Sorry for the newbie question (new to GitHub and Arduino). Does your comment mean "Cool idea, go ahead and implement that in your [me] own version," or "Good idea. I [Arduino developers] will make that change"?
```
@tjayrush  there may not be a straight answer ... i would not dare presume what the Arduino team will do. However you may want to read all the other open issues and rich your own conclusions about what they are likely/unlikely to implement. Speaking in my own name, i find the idea interesting and there is a chance i will do something about it in a [different setting](https://github.com/lmihalkovic/Arduino/issues/1)

VioletGiraffe, 2017-02-24:
I want to bump this issue. Detecting file changes and prompting to reload modified files is basic functionality in any IDE or even a plain text editor. Even if you don't do any fancy stuff like working on the same project in two different IDEs at the same time (which I do), you will still come across this problem as soon as you put your project under version control. You revert some changes or switch to another revision - boom, you have stale code in your IDE, possibly without noticing a problem for some time.

facchinm, 2017-02-24:
Unless you select "Use external editor", the IDE keeps the sketch is a memory buffer and saves it to a temporary file when needed. 
This allows compiling (and testing) a sketch without necessarily overwriting the original one (I doubt any other IDE does that, but it's very common to open Blink, change the pin and upload the code without saving anything). But since it's not disk-backed, this flow doesn't allow the sketch to reload a file while running as editor. 
If someone has a proposal that preserves the current behaviour while allowing a sketch to be reloaded I'll be glad to analyse and implement it.

VioletGiraffe, 2017-02-24:
I see what you mean, however it doesn't seem to be much different from any other IDE. You open files and edit them all the time so the files are in an unsaved state. It is not uncommon then to change the file externally, and when you go back to the IDE you get a prompt:

[image]

[image]

[image]

I don't see why the exact same behavior won't work with Arduino - without changing anything else.

facchinm, 2017-02-24:
The difference is that if you don't save a file in a normal IDE you compile the on-disk file, not the temporary copy. So for a normal IDE the "reload" dialog is a must, while in our case it would be somehow counter-intuitive. I'm not saying we are not going to implement somethig similar sooner or later but I'd give a low priority to this task

VioletGiraffe, 2017-02-24:
@facchinm : in my opinion that changes nothing and there's no problem here. Say you've opened some example, then edited and uploaded it. How probable do you think it is that the example file you started with will be modified? I think that's unlikely. In any of the cases where Arduino IDE's in-memory compilation feature is useful, I don't think a prompt I'm asking about would ever appear at all. It would appear in the cases where it's actually required (e. g. you're working with your sketch in a version control repository).

facchinm, 2017-02-24:
That makes sense, indeed, I'm only saying it's not top priority, and neither it's an easy check between (local) file timestamps

lmihalkovic, 2017-02-24:
This has been standard behavior in IDEs for years now.. the arduino environment persists in redifining what is standard expectation.. if you convince yourselves it is great.. but don t expect that it might always work with everyone

VioletGiraffe, 2017-02-24:
@facchinm:  I would do it with a file system watcher which is the OS-level tool just for this task, rather than by relying on timestamps.

lmihalkovic, 2017-02-24:
Not wanting to deal with the technical problems is one thing (we already know we are in different places on the continuum of what is 'difficulty'), but wanting to convince people they don t need what they wish for or .. is a very different type of response

lmihalkovic, 2017-02-24:
.. in my very humble opinion.

matthijskooijman, 2017-02-24:
I'd also like to see some prompt to reload on change, seems like a worthwhile addition to me (and has been on my mental TODO-list for a while, though I'll probably never find time for it). In my workflow I often use an external editor, but sometimes I quickly load a sketch and want to make some changes to it in the Arduino IDE. Now I end up toggling the external editor option every now and then, but that's not ideal. I also don't think that this would conflict with how the editor handles files currently.

Wrt implementation, the screenshots seem like good examples. I'd imagine something like:
 - When Arduino gets the focus, show a prompt for each file that has changed since it was last loaded or saved in the IDE.
 - It should have buttons to load the file from disk, not load the file from disk and probably also to apply the same answer to all files.
 - If the file is modified inside the Arduino IDE, the prompt should make this clear, and emphasize that loading from disk will throw away changes. Perhaps the "do not load" button should have a different text (such "Discard my changes").
 - When the file is not loaded from disk, it should be marked as modified in the IDE, and subsequently clicking save in the IDE will overwrite it. The IDE could check all files again just before saving and ask for confirmation saying something like "Files were changed on disk, you will overwrite these changes now".
- Detecting whether a file was modified should probably happen using the modified time of the file, perhaps combined with a checksum of the file content most recently saved or loaded.

supuflounder, 2017-07-06:
What disturbs me is that I see essential ideas posed years ago, such as auto-reload and NEVER a read-only IDE (I can fix trivial errors in the IDE but I want my fully-trained-to-me C++ editor for most program construction), compiler options, etc., but in many years nothing seems to have been done to make any of this happen.  The value of open-source is that we should be able to do all this.  Arduino Create is cute, but it is NOT a replacement for a full IDE.

I want a link map.  I need to set certain optimization options.  This is essential.  I have used IDEs of various vendors, and open source (e.g., Eclipse) for decades, and I have never seen an IDE as gratuitously restrictive as the Arduino IDE.  And there is no excuse for it.  It should not take four years to get -D options into the IDE, or configuration options, or having an automatic build for two different platforms (select one, or two, or all).  And the business of always recompiling on a download, even if no changes have been made, is just plain silly.  A download should only recompile if a module has changed since the verify.

This limited IDE is impacting my ability to get work done.

Why is nothing being done? 
     joe

matthijskooijman, 2017-07-06:
> Why is nothing being done?

A lot of things are being done, but apparently just not the things you care about mostly. There's just not enough time to implement everything right away. I won't go into detail about the other issues you raise, let's not clutter this discussion with meta-discussion.

As for this particular issue, it's still on my wish-list, but not quite at the top (yet). If you care about this particular feature, feel free to implement it and submit a PR, *that's* Open Source. If you do, I'll try to find some time to review the changes and get them in mergeable shape.

per1234, 2017-07-07:
It should be noted that in fact something is being done about auto reload! See https://github.com/arduino/Arduino/issues/6074. @supuflounder, if you want to help get this in the next release feel free to install the test build from https://github.com/arduino/Arduino/pull/6074#issuecomment-312810714, try it out, and comment on that pull request with your results. You'll see the milestone was set for 1.8.3 but that ended but being a minor bugfix service release so I think there is a good chance of it being in 1.8.4.

supuflounder, 2017-07-07:
As long as I know the path, I'm willing to undertake making the mods I
want.  Would you have any problem if I added a couple of my favorite
editor features, disabled by default?
     Joe

samdenty99, 2017-07-19:
This would be useful as I use scripts that auto-generate constants in my sketch. Implementing this would be fairly simple by checking the file hashes on window focus, as well as saving a nearly a minute to close and re-open the sketch

heX16, 2018-02-25:
This is really a very necessary feature! I've just put the registers in order for two hours (using an external editor). Then (before the commit) I decided to check the changes - pour them into the controller. And when I did the controller firmware I lost everything! Because in the open IDE I unintentionally first pressed "U", and then "ctrl+U" - stupid IDE save file and wipe my works.

facchinm, 2018-03-02:
@heX16 the Beta branch implements a watcher on external files, you could give it a try and report if it works to speedup inclusion :wink: Otherwise, selecting "Use external editor" in Preferences will avoid overwriting anything in currently released IDE versions.

supuflounder, 2018-03-04:
> @heX16 the Beta branch implements a watcher on external files, you could
> give it a try and report if it works to speedup inclusion :wink:
> Otherwise, selecting "Use external editor" in Preferences [...]

If you record a timestamp with every change (and reset the timestamp on
undo), and record the timestamp of the file that loaded the buffer, we
have the following states

buffer load timestamp == file timestamp:  save valid from IDE
unchanged buffer load timestamp  < file timestamp:  when activated, reload
the buffer from the file
changed buffer load timestamp < file timestamp: when activated, issue
error message indicating sync error
changed buffer timestamp > file timestamp, buffer load timestamp == file
timestamp.  Good

A request to save (not save as) an unchanged buffer should be ignored.
There ahould be an option to Save All, which saves modified buffers
(subject to the above limitations) and leaves unmodified buffers alonr

Im not sure this list is comprehensive; I implemented this over 20 years
ago.
      joe

Humancell, 2019-06-18:
I love to see that this feature has been worked on, and seems to have a series of commits related to it, but I'm guessing that right now this has not made it into any Release yet?

facchinm, 2019-06-19:
@Humancell you can test the feature on the Beta branch of the IDE (download links are here https://www.arduino.cc/en/Main/Software#beta ).
Let us know if it works for you so it would speedup merge into master :wink:

Humancell, 2019-06-23:
Well, I pulled down and installed arduino-PR-beta1.9-BUILD-107 on Windows.

I went through the following steps:

1. Created a new project and saved it.  Left the IDE open.
2. Opened the .ino in my external editor.  (Notepad++) Made a change and saved.
3. The IDE did not do anything.  It did not seem to detect the file change underneath it.
4. I edited the file in the IDE, adding a comment, and saved.  It overwrote the changes from Notepad++.
5. When I returned focus to Notepad++ it immediately indicated that the file was changed under it.

What else would you like me to test or report?

facchinm, 2019-06-25:
Indeed, it looks like the hot reloading is not working in your setup. Did you get any Java error on the "Console" ?

commit 7640adb5 (2018-01-23): Add watcher on sketch files

Reloads sketch content if the Editor is not in foreground and isomething happens in the backing storage files.

Note that no confirmation dialog is displayed (same behaviour as SublimeText, differs from other IDEs)

Fixes #4551 and #5345
commit ab1e13df (2018-05-14): Add watcher on sketch files

Reloads sketch content if the Editor is not in foreground and isomething happens in the backing storage files.

Note that no confirmation dialog is displayed (same behaviour as SublimeText, differs from other IDEs)

Fixes #4551 and #5345
commit e29e3a30 (2018-05-15): Add watcher on sketch files

Reloads sketch content if the Editor is not in foreground and isomething happens in the backing storage files.

Note that no confirmation dialog is displayed (same behaviour as SublimeText, differs from other IDEs)

Fixes #4551 and #5345
commit 354b914f (2018-05-22): Add watcher on sketch files

Reloads sketch content if the Editor is not in foreground and isomething happens in the backing storage files.

Note that no confirmation dialog is displayed (same behaviour as SublimeText, differs from other IDEs)

Fixes #4551 and #5345