arduino/Arduino #4552: Windows installer will require updated SHA2 signature

PaulStoffregen, 2016-02-10:
The next release of Arduino will required updated SHA2 signing on Windows.

Failure to update Arduino's code signing certificate and build process before release of Arduino 1.6.8 will result in Windows 7,8,10 users getting a red error message: signature of this file is corrupt or invalid.

Here is Microsoft's not-terribly-useful security advisory about this change.

https://technet.microsoft.com/library/security/3123479

Programs built before Jan 1, 2016 are "grandfathered" until 2017 from this check, which is why Arduino 1.6.7 and prior releases have not yet experienced this problem.  Anything built after Jan 1, 2016 (according to the cryptographically certified timestamp) requires SHA2 to avoid these alarming messages.

Arduino needs to first get an updated code signing certificate.  The certificate used can be checked by right-clicking on the installer and choosing properties, then the Digital Signatures tab, then Details, then View View Certificate, and then the Details tab, which shows the certificate's signature algorithm.

(screenshot)

After you've got a certificate using SHA2, then you need to update Arduino's build process to generate a SHA2 signature.  The main properties dialog shows which algorithm you used to sign.

(screenshot)

Both your signature and the signature within your code signing certificate must be SHA2 to pass the new checks in Windows 7,8,10.

I know this in unpleasant news.  But in case this wasn't already on your radar, it's far less painful to start the process of getting your code signing certificate re-issued and updating build settings *before* releasing an installer which alarms users with an invalid signature.

More info can be found here, and many other sites.

http://support.ksoftware.net/support/solutions/articles/215805-the-truth-about-sha1-sha256-and-code-signing-certificates-

https://support.comodo.com/index.php?/Default/Knowledgebase/Article/View/1102/7/code-signing-certificates---sha1-and-sha-256-information

facchinm, 2016-02-10:
Hi Paul,
thanks a lot for the very clear infos and analysis. 
But I'd want to point out that we are using the same key to sign the Create agent, and since the installer is recreated in continuous integration it should already be affected by this "bug".
Would you mind testing the `Windows dev` installer [here](https://github.com/arduino/arduino-create-agent#contributing) in your environment? In my W10 VM there is no scary message when downloading/installing.
Anyway, next releases will for sure be signed with a sha2 certificate :smile:

PaulStoffregen, 2016-02-10:
I'm afraid it's not looking so good....

PaulStoffregen, 2016-02-10:
[image]

PaulStoffregen, 2016-02-10:
As you can see in these screenshots, checking your file is as easy as running Internet Explorer on any up-to-date Windows 7,8,10 machine.  Just download the file, and see if those red "signature of this file is corrupt or invalid" messages appear.

Admittedly, this does involve using both Windows and Internet Explorer briefly... something I personally don't find so easy.

PaulStoffregen, 2016-02-10:
At least better to find out now and update your cert and scripts *before* releasing to a wide audience!

awatterott, 2016-02-11:
Here is a really good guide about Windows code + driver signing from David Grayson:
http://www.davidegrayson.com/signing/
As digest algorithm (file hash) you can still use SHA-1.

tonyrsutton, 2017-01-16:
Bumping this up... this expires in 2 days time for the "Arduino LLC" security certificate.

facchinm, 2017-01-16:
@tonyrsutton we'll be signing the upcoming releases with a new (not expired) certificate, however all IDEs from 1.6.9 on are double signed with SHA1 and SHA256 (for extended compatibility with WinXP and Win7 SP0).

closed 2017-01-16 by facchinm

tonyrsutton, 2017-01-16:
Thanks. Can you please confirm if v1.8.1 has the certificate time stamped, so the install will still go ahead after 18th Jan? (especially these drivers)

Cheers,

facchinm, 2017-01-16:
Confirmed, both installer and drivers are timestamped and won't decay in 2 days :smile:

tonyrsutton, 2017-01-16:
Excellent, many thanks for your replies.