arduino/Arduino #4558: Problems with Arduino IDE 1.6.7

chessweb01, 2016-02-11:
I have two issues with Arduino IDE 1.6.7 that have not been present with 1.6.5:

-   When I enter the boards manager, I get the error message: package_index.json file signature verification failed

-  Compilation of sketches for Arduino Due (actually the clone SainSmart Due) fails with the following message:
Arduino: 1.6.7 (Windows 7), TD: 1.27, Board: "Arduino Due (Programming Port)"
Board arduino_due_x_dbg (platform sam, package arduino) is unknown
Error compiling.

The boards manager shows that Arduino SAM boards (32 Bit Arm-Cortex M3) by Arduino Version 1.6.5 are installed, but due to the first problem I cannot try to solve the the second one by upgrading to 1.6.

facchinm, 2016-02-11:
HI @chessweb01,
this happens because there are two (or zero) versions of `sam` core in your installation and the builder get confused.
I can suggest you to delete `C:\\yourUsername\AppData\Local\Arduino15\packages\arduino\sam` and `C:\\yourUsername\AppData\Local\Arduino15\package_index.json`, and also `C:\\yourUsername\AppData\Roaming\Arduino15\` if you have it in your system.
Then closing and reopening the IDE should solve the issue.

facchinm, 2016-02-11:
For reference: https://github.com/arduino/arduino-builder/pull/84

chessweb01, 2016-02-11:
Hi @facchinm, thanks for the quick response, but that didn't change anything.

First of all, I don't have `C:\\yourUsername\AppData\Local\Arduino15\packages\arduino\sam` folder, but rather a `C:\\yourUsername\AppData\Local\Arduino15\packages\arduino\**hardware**\sam` folder which I renamed. Deleting the other folder and the json-file didn't help either.

facchinm, 2016-02-11:
Yes, sorry, I meant `C:\\yourUsername\AppData\Local\Arduino15\packages\arduino\hardware\sam`
You should have it, at least to show `Arduino Due` in the `Boards` menu

chessweb01, 2016-02-11:
Yes, but as I said deleting it didn't change a thing. Still the same errors.

facchinm, 2016-02-11:
Just to make sure, did you restart the IDE after deleting the folder? You shouldn't have any `Arduino Due` entry in the menu after restart, so no way to select it and cause the `Board arduino_due_x_dbg (platform sam, package arduino) is unknown` error

chessweb01, 2016-02-11:
I went ahead and uninstalled Arduino IDE. After removing everything Arduino related from  ...\AppData\... I reinstalled Arduino IDE 1.6.7. That got rid of all my problems :-)

facchinm, 2016-02-11:
A bit rough but functional :wink: 
I'm going to close this issue as "solved, don't know what happened" :smile:

closed 2016-02-11 by facchinm