arduino/Arduino #4561: Does not contain a Windows JVM

iHateInventNames, 2016-02-11:
Trying build on Gentoo Linux:
BUILD FAILED
/home/name/sources/oo/Arduino/build/build.xml:110: The following error occurred while executing this line:
/home/name/sources/oo/Arduino/build/build.xml:87: The following error occurred while executing this line:
/home/name/sources/oo/Arduino/build/build.xml:924: It looks like /usr/lib64/icedtea8/jre does not contain a Windows JVM

facchinm, 2016-02-12:
Hi,
do you need to compile the IDE to run locally or to package it?
In the first case, once setting up JAVA_HOME env variable to point to Java8 (Oracle or openjdk), you can simply execute
```bash
cd build
ant build run
```
In the second case, you need to pass a couple of defines. For example, when packaging for Linux64, you should execute
```
ant dist -Djava.net.preferIPv4Stack=true -Dplatform=linux64 -DLINUX64_BUNDLED_JVM=/opt/jvm/linux64_jre1.8.0/
```

iHateInventNames, 2016-02-23:
Thank you much

sandeepmistry, 2016-03-09:
This looks resolved, closing.

closed 2016-03-09 by sandeepmistry

iHateInventNames, 2016-03-11:
I want to make a fresh ebuild for Gentoo