arduino/Arduino #4562: add a bit of space around the left-hand rule

RichMorin, 2016-02-11:
I think the IDE would look better and be a bit easier to read if a couple of white pixels were added on each side of the left-hand rule (just to the left of the first character in the code line.

facchinm, 2016-02-12:
@00alis

00alis, 2016-02-12:
agreed!

per1234, 2016-02-12:
I always have trouble when selecting text with the mouse in the Arduino IDE editor because when I move the mouse pointer over the left margin trying to get the leftmost character it no longer can select text. I use TextPad sometimes and I really like the (optional?) fairly large left margin that still allows for text select. So maybe increasing the space could also help with this issue.

CarlosMiguelCuevas, 2016-03-24:
I don't know if I got your ideas correctly, but I'm submitting a change in the code to modify that, if someone of you could verify if that's what you want, that would be great. 
Pull Request #4750