arduino/Arduino #4569: [Improvement] - Add option to enable/disable network/serial discovery

docbender, 2016-02-12:
Would be nice to have option in preferences to control network and serial port discovery (option to disable it).

In my case (Win7 64bit Java 8u73 and 5 network cards) discovery extremely stress CPU. One core usage is up 100% in 3-second intervals. This load is dependent on the number of allowed network interfaces a lot.

facchinm, 2016-02-15:
Hi @docbender ,
the high cpu usage for Serial discovery has been solved in mainline (you can test using the latest hourly version). Network discovery code is still untouched but I never saw it causing a huge performance drop (I must admit I've never used a PC with 5 network interfaces :smile: )
If you could report if the hourly solves your problems (at least partially) it would be great!

docbender, 2016-02-15:
Hi, I tested hourly version from wednesday. I can try latest one and put here some CPU usage screenshot.

Probably I am not only one who saw this kind of issue:
https://forum.arduino.cc/index.php?topic=347870.0

facchinm, 2016-02-15:
@docbender , could you test [this](https://github.com/arduino/Arduino/pull/4482#issuecomment-184294064) version of the IDE?
I've stripped down the network discovery code to leave the barebone functionalities only (with a 5 second timer task to check for the board reachability)

docbender, 2016-02-15:
@facchinm , I tested your IDE version and CPU usage rapidly dropped down. 

I attached two images - CPU usage screenshots.  First belong to latest nighly build and second to your version. (my CPU is 4 core so 25% is one thread fully utilization)

[image]
[image]

sandeepmistry, 2016-07-18:
Closing this, now the the high CPU issue has been resolved, I don't think this request is needed any more.

facchinm, 2016-07-19:
The mdns changes were not merged at the end since there were problems with multiple network ports. Leaving this open for reference if sooner or later we want to tackle that issue