arduino/Arduino #4572: Theme colors issue since 1.6.7  by LITERAL1

MarkusLange, 2016-02-13:
In IDE 1.6.7 the Color of the LITERAL1 tagged Constants from the keywords.txt are shown orange not blue.

Since IDE 1.6.4 (I looked back from 1.6.7 the LITERAL1 is shown orange not blue) is the color from LITERAL1 nocht correct inside the editor.

facchinm, 2016-02-15:
@00alis

PaulStoffregen, 2016-02-15:
I've noticed this too, in several libraries I maintain which use LITERAL1.

00alis, 2016-02-16:
You are right! Thank you for reporting this!
We are going through a review of the Arduino pigmentation in the next few days and we will add this bugfix. @agdl

agdl, 2016-02-19:
it should be ok now with https://github.com/arduino/Arduino/commit/88650b65a699dcddbe0ac58bbabd9b19cf882c1e

MarkusLange, 2016-02-19:
Changed the files to 88650b6 in 1.6.7 but no luck

agdl, 2016-02-22:
Try now please this time should be really ok with https://github.com/arduino/Arduino/commit/8035f244989f92830bd5a72f5a7d8d9f3aa7f6c8

If so close the issue please

MarkusLange, 2016-02-22:
May I did something wrong but it doesn't help, sorry.

PaulStoffregen, 2016-02-27:
I can confirm the problem is NOT FIXED in the latest nightly build.

PaulStoffregen, 2016-02-27:
As you can see, a library using LITERAL1 isn't getting the proper blue color.

To reproduce the problem, install the Encoder library, and then add "#define ENCODER_USE_INTERRUPTS".

agdl, 2016-02-27:
@PaulStoffregen yes I know. I understood where the problem is. On Monday i will fix it

agdl, 2016-02-27:
If you switch to the branch I created and referenced in this issue. Looking at the files I modified you can easily see that working on that it is possible to highlight many different keywords. It will be very useful for future developments.

PaulStoffregen, 2016-02-28:
I just tried the themeKeywords branch, compiling from source on Ubuntu 14.04 (64 bit).

The colors do not seem to be correct....

agdl, 2016-02-28:
@PaulStoffregen yes it is a work in progress we have to decide colors with @00alis

commit 057bf639 (2016-02-29): This fixes #4572
closed 2016-02-29 by agdl