arduino/Arduino #4575: Remove or update the arduino and arduino-core packages on Ubuntu's repository

Avamander, 2016-02-14:
It would be really nice if people could automatically update their Arduino IDE with their package manager automatically but currently the package available from Ubuntu's repository is 1.0.5 would it be possible to start updating it again? Offering different channels like stable/unstable would be even nicer.

per1234, 2016-02-15:
I believe what is preventing the recent versions from being available is https://github.com/arduino/Arduino/pull/2703

cousteaulecommandant, 2016-02-16:
Ubuntu inherits most of its packages from Debian, and Debian's Arduino is 1.0.5.  However the `experimental` branch of Debian has 1.5.6.2 (wow, so much new).  Probably the way to solve this is to convince Debian first to update Arduino, and then insist Ubuntu to "inherit" it ASAP.  (This is, if there are no license issues.)

Alternatively, Areuino could provide a PPA if it doesn't already.

Avamander, 2016-02-16:
Not Debian needs to be convinced, someone dealing with the Arduino project
needs to.

matthijskooijman, 2016-02-17:
I'm closing this issue, at is indeed something to report upstream. AFAIK the update in Debian is indeed awaiting license clarification, but that is the subject of #2703, so there is no point in keeping this open separately as well.

closed 2016-02-17 by matthijskooijman