arduino/Arduino #4584: String class not behaving as expected with Print class on M0+

tigoe, 2016-02-17:
Initializing a String object with a float and then printing it fails on the M0+ platform.

Try this:

`
void setup() {
Serial.begin(9600);
}

void loop() {
  float foo = 3.14159269;
  String fee = String(foo, 2);
  Serial.print(foo, 5);
  Serial.print(" ");
  Serial.println(fee);
}
`

This should produce

3.14159 3.14

Instead it only produces 
3.14159

facchinm, 2016-02-17:
Hi Tom,
this is a know issue, tracked as https://github.com/arduino/ArduinoCore-samd/issues/109 and probably solved by https://github.com/arduino/ArduinoCore-samd/pull/111
Closing the issue here and keeping track on `samd` core repo

closed 2016-02-17 by facchinm

tigoe, 2016-02-17:
Excellent, thanks!