arduino/Arduino #4585: Importing a library created recursive folders...

javikalback, 2016-02-17:
When trying to import a new library I double clicked on a folder that contained the arduino libraries by mistake. This cause some kind of weird situation in which the folder copied recursively and not it has "no end" it will not let me delete or rename because the path is too long. I have tried to rename some folder manually just to find more and more folders inside. It is really never ending! I've tried several tools i found online as well as system commands workarounds such as robocopy but I've had no luck removing this silly folder (and all its children).

This should be looked into so no one else does the same stupid thing I did. I am thinking my last resort is to format my computer :(

Chris--A, 2016-02-17:
What IDE version?

This was supposed to be resolved: https://github.com/arduino/Arduino/issues/2795

javikalback, 2016-02-17:
I am not sure... I uninstalled the ide hoping it would delete itself.
Spoiler alert: it didn't work. I know I installed it about a month ago, I
will try to see if I can find the exact version somehow.

Does anyone know how to remove the "infinite" folder? I've tried different
things with no luck :(

Javi Kalback

javikalback, 2016-02-18:
Hi @Chris--A
I searched my download folder and looks like the version I had installed is 1.7.8  

cc: @cmaglie

agdl, 2016-02-18:
@javikalback I'm sorry but you downloaded an IDE that is not developed by us, so we can't help you, you have to open an issue on their repository. Our IDE can be downloaded here
https://www.arduino.cc/en/Main/Software

closed 2016-02-18 by agdl