arduino/Arduino #4591: Running couple times and cannot upload the code again

LexiLam, 2016-02-18:
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:140)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2430)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.SerialException: Error touching serial port '/dev/cu.usbmodem1411'.
	at processing.app.Serial.touchForCDCReset(Serial.java:87)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:129)
	... 6 more
Caused by: jssc.SerialPortException: Port name - /dev/cu.usbmodem1411; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.touchForCDCReset(Serial.java:81)
	... 7 more

facchinm, 2016-02-18:
Hi @LexiLam ,
what version of the IDE are you using?

LexiLam, 2016-02-18:
Yun and Uno, arduino 1.6.7

facchinm, 2016-02-23:
The issue should be solved by 2893c2d64314f3fa298fffb5c90d8e27c8614d08, released just after 1.6.7.
Can you test the hourly build and report if you are able to reproduce it?

sandeepmistry, 2016-07-18:
Closing this due to lack of feedback.

closed 2016-07-18 by sandeepmistry