arduino/Arduino #4594: SMlib does not run the next Head()

dangu, 2016-02-19:
I downloaded the SMlib from here: http://playground.arduino.cc/Code/SMlib

When running some tests I noticed stange behaviour when a Set() was run inside the current state. The next Head() state was then not run, instead it continued on to the next Body(). Maybe this is best explained by an example:
SM MainSM(S1_head, S1_base);

State S1_head(){//Do something}

State S1_base()
{
MainSM.Set(test_one); // Here the next state is set to test_one
}

State test_one()
{
	MainSM.Set(S1_head, S1_base); // Here S1_head should be run the next exec BUT it never will,
                                      // as if(!Finished) Exec = &SM::Run; will overwrite FRun (in SM.h)
}

I could not find any github-codebase nor a maintainer to this library. Should I upload my own version to GitHub or how could this be fixed?

My patch looks like this:
--- a/SM.h
+++ b/SM.h
@@ -31,8 +31,8 @@ typedef void(SM::*PRun)();//defines runstate pointer
     };//Run()
     void FRun(){//runstate entry
       Mark = millis();//reset timer on entry
-      Head();//call head state function
       if(!Finished) Exec = &SM::Run;//set runstate to running
+      Head();//call head state function
     };//FRun()
     void Idle();//runstate finished
     unsigned long Mark;//timer mark

per1234, 2016-02-19:
If you look at the [history](http://playground.arduino.cc/Code/SMlib?action=diff) for that playground page you can see the author's arduino.cc user name is [nilton61](http://forum.arduino.cc/index.php?action=profile;u=170491). There was another issue with this library of the examples not compiling with Arduino IDE 1.6.6+ reported recently https://github.com/arduino/Arduino/issues/4537 and I sent the author a PM on arduino.cc notifying them of the required changes but I haven't heard back yet and their [last forum post](http://forum.arduino.cc/index.php?topic=313442.msg2442532#msg2442532) indicates they may no longer be using Arduino. Regular users are no longer allowed to upload files to the Playground but the webmaster did offer to add the updated file if requested. A GitHub repository would be much better because then we could just submit pull requests or issues to it.

dangu, 2016-02-19:
Ok. Now I put my code here: [https://github.com/dangu/SMlib](https://github.com/dangu/SMlib)
If nilton61 wants to, I would be happy to transfer the ownership of this repo.

dangu, 2016-02-22:
The fix is now released as v1.3 and I have updated the page with links to this version here: http://playground.arduino.cc/Code/SMlib

closed 2016-02-22 by dangu