arduino/Arduino #4595: Support for ATtiny2313A

czesiu, 2016-02-19:
Hi,

I want to add support for ATtiny2313A to damellis's attiny package (https://github.com/czesiu/attiny/commits/ide-1.6.x-boards-manager).

I'm using Arduino 1.6.7.
When I set in boards.txt file mcu as:
`attiny.menu.cpu.attiny2313.build.mcu=attiny2313`
then during compilation the following error occurs:
`~\AppData\Local\Arduino15\packages\attiny\hardware\avr\1.0.2\variants\tiny20x13/pins_arduino.h:71:48: error: 'PCIE0' was not declared in this scope`

When I set in boards.txt file mcu as:
`attiny.menu.cpu.attiny2313.build.mcu=attiny2313a`
then code successfully compile, but I got following error:
`avrdude: AVR Part "attiny2313a" not found.`

Temporarly I fixed problem by including avr/iotn2313a.h instead of avr/iotn2313.h in io.h file (line 414).

Could you repair that in official release?

agdl, 2016-02-26:
@damellis

wasimxe, 2016-05-06:
hi, do anyone have boards.txt and *.hex for attiny2313A? 
could any one please help bootloading attiny2313A 8Mhz internal RC Oscillator, here is error log file http://pastebin.com/PZvWsV3P and boards.txt http://pastebin.com/z4He7psp

czesiu, 2016-05-09:
@wasimxe, you can add [this](https://raw.githubusercontent.com/czesiu/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json) to board manager. Then you can choose Attiny2313A, but you should change include ```avr/iotn2313.h``` to ```avr/iotn2313a.h``` in ```<arduino-dir>/hardware/tools/avr/avr/include/avr/io.h``` file (line 414).

wasimxe, 2016-05-10:
@czesiu  thanks for help, also this URL help me and I successfully bootloaded and upload sketch 
https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md

agdl, 2016-07-07:
I'm closing this issue. Since the Tiny core is not part of the arduino product line apart for the Gemma.

closed 2016-07-07 by agdl

gulabartosz, 2020-10-30:
Thanks @czesiu!