arduino/Arduino #4604: SMlib code is copied to github

dangu, 2016-02-22:
Hello

I have taken the code from the library [SMlib](http://playground.arduino.cc/Code/SMlib) and put it on github: https://github.com/dangu/SMlib

This way the code can more easily maintained. Please add it to the Library Manager, and tell me if there is something more to be done to make it conform to the Arduino library standard.

Also, this library was once written by @nilton61. I did not want to steal their code, only make it more maintainable. So I would be happy to transfer the ownership to @nilton61 if requested for.

nilton61, 2016-02-22:
Excuse me???
I have NOT given my consent to this

dangu, 2016-02-22:
Ok. Then please tell me what to do. Should I remove the repo from github?

dangu, 2016-02-22:
Or, I will just transfer the ownership to you, and you decide what you would like to do with it?

nilton61, 2016-02-22:
Why did you do that in the fist place? What kind os a person are you that
does not ask permission BEFORE taking such a step?

dangu, 2016-02-22:
I only had good intent with all of this, just to help with the development of this library. Reading the comments of https://github.com/arduino/Arduino/issues/4537 and https://github.com/arduino/Arduino/issues/4594 I saw it as a welcome step to move the development forward. I am very sorry to have insulted you, which, again, was not my intention.

matthijskooijman, 2016-02-22:
@dangu, it would probably have been better (and looking at it from a legal perspective, required) to contact @nilton61 beforehand, though it is not quite uncommon for code to be dumped on the playground without looking back.

@nilton61, it seems dan

nilton61, 2016-02-22:
I do not question your intent. But for me it is UNTHINKABLE to act like you
did. And even the fact that the idea occurred is very strange to me. So
please delete the code from github and restore it to playground if
necessary and never contact me again.
And fyi: I would gladly have given my permission if that was requested
beforehand.

matthijskooijman, 2016-02-22:
@nilton61, I just submitted my comment while it was halfway finished. I've edited in the remaining remarks, which I'll repeat here for your email-notification convenience:

@nilton61, it seems dangu is more than willing to move forward in any way you see fit, so it would seem the most constructive if you would answer his questions about how you want to move forward. Keeping a github repository seems like a good idea to me, and is even required to distribute the library through Arduino's library manager.

Also, I strongly recommend you pick a license for your code, since without a license your code is pretty much unusable. See also this blogpost for some more info on how this works: http://blog.codinghorror.com/pick-a-license-any-license/

matthijskooijman, 2016-02-22:
@dangu, you might also want to check out the blogpost I linked, since it explains a bit how copyright treats things without an explicit license. Perhaps you were under the impression that code published on the internet / playground was intended for others to use and it would thus be acceptable to create a modified / fixed version of such code, but as that post explains, in absence of any license, copyright law prevents you from doing exactly that.

dangu, 2016-02-22:
Status: I have now removed the github repository of SMlib. I have also restored the page http://playground.arduino.cc/Code/SMlib as it was.

As this is my first time trying to contribute to the Arduino community I am very sorry that it ended up this way. I only thought I was helping the development forward.

@matthijskooijman: You are right. My understanding was that the playground-code was there to use and modify.

nilton61, 2016-02-22:
I dont mind people using my code without referring to me, i dont mind
people changing my code either actually i do not care very much about what
happens to my code. if anybody seems fit to maintain it and develop it,
fine for me.
But i do mind very much the way this was done. And that made me decide not
to be constructive in any way at all, in the contrary, i will be as
obstructive as humanly possible after this, sorry

nilton61, 2016-02-22:
Yes, but ask first!! What planet are you from??

2016-02-22 16:50 GMT+01:00 nilton61 . <nilton61@gmail.com>:

cmaglie, 2016-02-22:
Ok, I'm closing this one, since it seems that there is nothing more to say.

@nilton61 if you ever decide to publish the library, with your own pace, you're welcome of course.

closed 2016-02-22 by cmaglie

nilton61, 2016-02-22:
No, i will delete it from the playground as well

agdl, 2016-02-22:
@nilton61 good to know thank you, for the demonstration of tolerance and community based sharing! :+1:

nilton61, 2016-02-22:
Before this happened i have been very happy to share.
And yes. I have NO TOLERANCE AT ALL for the kind of behavior exhibited
here. And frankly speaking nobody should, because any tolerance will be
interepreted as encouragement.
And your last comment made make a decision not to continue sharing my code
or my ideas at all. Hope your happy now

Michael-Bell, 2016-06-06:
@nilton61 

I don't want to stir up any bad feelings, or to step on any toes, but I would like to clarify if you have any plans to reshare your code in the future. I'd really like to use SMLib for my projects, as I have been, but I don't want to use the code if it is no longer licensed for use. 

If you are willing to allow use, can you re-upload the code to the playground with a license so that new users are able to obtain the code, as most google results for Arduino state machines suggest your code is the best for a simple implementation, and a lot of users would be directed to an empty page. 

Otherwise, is it possible to leave a note on the Playground SMLib page suggesting possible alternatives to users so they are not left feeling discouraged?

Again I don't want to cause anymore drama, I just thought I'd ask.

nilton61, 2016-06-07:
I am sorry but i have no intention of spending one single second on this
any more, and that includes writing replies like this

Chris--A, 2016-06-07:
@Michael-Bell 

Here is a link to an archive of the playground with the zip file of the code... However remember the code is unlicensed. The link @matthijskooijman posted explains it well 

"_People can read the code, but they have no legal right to use it. To use the code, you must contact the author directly and ask permission._"

https://web.archive.org/web/20150723075012/http://playground.arduino.cc/Code/SMlib

matthijskooijman, 2016-06-07:
@Chris--A, AFAIU, it's subtly different: Using code you obtained from the author without a license is actually allowed according to most copyright laws, but *distributing* a copyrighted work without permission is forbidden (almost?) everywhere. In this case, this means archive.org might be in violation of copyright (though I'm not sure if there are exceptions for archives like these, or what their policy is for removing content on request of the author).