arduino/Arduino #4606: Retrieve current pin mode

mikaelpatel, 2016-02-23:
There is currently no function to get the current pin mode. A possible implementation is:
```
int pinMode(uint8_t pin)
{
  if (pin >= NUM_DIGITAL_PINS) return (-1);

  uint8_t bit = digitalPinToBitMask(pin);
  uint8_t port = digitalPinToPort(pin);
  volatile uint8_t *reg = portModeRegister(port);
  if (*reg & bit) return (OUTPUT);

  volatile uint8_t *out = portOutputRegister(port);
  return ((*out & bit) ? INPUT_PULLUP : INPUT);
}
```
The return value is the pin mode or error code (-1) for illegal pin number.

Cheers!

liamkinne, 2017-12-30:
Only negative I can think of is the fact that for some variants the return type of portModeRegister() is of type `volatile uint16_t *`. On personal projects I have used the `auto` type specifier to fix this, but I don't know if Arduino is using c++11 by default. For now change I suggest changing it to `uint16_t` rather than `uint8_t`

Benjalien, 2023-01-30:
No news on this one? I'd love to have it native!
Thanks