arduino/Arduino #4608: 3rd party board [Tools -> Burn Bootloader] fail

Cheong2K, 2016-02-24:
Hi there,

I am making a new board, in boards.txt I added:
my_board.bootloader.tool=dfu

In platform.txt I added:
tools.dfu.erase.params.verbose=-v
tools.dfu.erase.params.quiet=-q -q
tools.dfu.erase.pattern="tool-erase"

tools.dfu.bootloader.params.verbose=-v
tools.dfu.bootloader.params.quiet=-q -q
tools.dfu.bootloader.pattern="tool-prog"

When I click, Tools -> Burn Bootloader, the following error will be shown:
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:314)
	at processing.app.Editor.lambda$handleBurnBootloader$42(Editor.java:2754)
	at java.lang.Thread.run(Thread.java:745)
Error while burning bootloader.

Cheong2K, 2016-02-24:
My board is in:
/Users/Cheong2K/Library/Arduino15/packages/my_company

However, it I move it to the arduino folder, it works just fine:
/Users/Cheong2K/Library/Arduino15/packages/arduino

sandeepmistry, 2016-02-25:
@Cheong2K a few suggestions:

Have you tried to add a programmer to ```programmers.txt```? (this is based off the SAMD [programmers.txt](https://github.com/arduino/ArduinoCore-samd/blob/master/programmers.txt))

```
my_company_dfu.name=DFU tool name
my_company_dfu.name.communication=USB
my_company_dfu.name.protocol=
my_company_dfu.name.program.protocol=
my_company_dfu.program.tool=dfu
my_company_dfu.program.extra_params=
```

Is your tool in ```~/Library/Arduino15/packages/my_company/tools/dfu```? I suggest adding a path like the SAMD core:

```
tools.dfu.path={runtime.tools.dfu.path}
```

Cheong2K, 2016-02-26:
@sandeepmistry,

It works just fine, thank you!!

closed 2016-02-26 by Cheong2K