arduino/Arduino #4609: Wrong error line indication for multi tab sketch

netmaniac, 2016-02-25:
I have encountered problem with Arduino IDE 1.6.7 (Linux 64 bit) showing wrong place where error during compilation has occurred. For me this is only case for error `void value not ignored as it ought to be`, other seems to be OK.

When sketch is in two or more tabs, and void function is used as non-void value in other tab when defined IDE returns error `void value not ignored as it ought to be` but it shows correct line number and *wrong* tab - where function is defined and not where it was used as non void value. Example in attached sketch. 

*Expected behavior* When compiling it should throw error and show line 3 in tab _not_ignored_void_
*Current behavior* Shows error in line 3 in tab _tab2_

This make very hard, especially for not advanced user to identify real problem with program.

Since I can not drop .zip file as attachment (Ubuntu 15.10 FF44.0.2) despite it should be supported example code is in this gist: https://gist.github.com/netmaniac/cbff2b5e6476f52a9d7c

per1234, 2016-02-25:
This issue has been previously reported at https://github.com/arduino/arduino-builder/issues/99.

Until the bug is fixed the workaround is to either manually declare the function prototypes for any function defined after the first tab, so you would change not_ignored_void.ino
 to:
```arduino
void add_two();  //function prototype for add_two()

void setup() {
  Serial.begin(9600);
  int val = add_two();
  Serial.println(val);
  
}

void loop() {
}
```
or you can use Arduino IDE 1.6.6(or previous) which doesn't have this bug.

per1234, 2017-07-04:
The error highlight on the demonstration code provided by netmaniac is now correct in Arduino IDE 1.8.3. I believe this was fixed by https://github.com/arduino/arduino-builder/pull/124.

closed 2017-07-04 by per1234