arduino/Arduino #4610: Add "Serial0" to keywords.txt in Adruino IDE

EmbeddedMan, 2016-02-25:
Just like Serial1, Serial2, etc. some boards (like FubarinoSD, DP32, and chipKIT Pi) sometimes use a serial object called Serial0. Including this in keywords.txt would make the use of Serial0 less confusing to users.

matthijskooijman, 2016-02-25:
I wonder if a platform has its own keyword.txt (or keywords.txt?), perhaps it would be better to put in there for those platforms, if this is possible?

PaulStoffregen, 2016-02-25:
Yes, 1.6.7 supports a custom keywords.txt for each platform.  It goes in the same location as boards.txt and platform.txt.

EmbeddedMan, 2016-02-25:
Woah! No way. That's simply awesome. A far better solution than changing
hard-coded values in the IDE. Thanks guys!

closed 2016-02-25 by EmbeddedMan