arduino/Arduino #4612: Arduino GSM library - can't change the TCP port

brezinapeto, 2016-02-25:
Please read communication with Quectel :

> Currently I work with Arduino GSM Shield with QUECTEL M10 chip. I use standard GSM library (https://www.arduino.cc/en/Reference/GSM) for communication with this shield. 
> From this point of view it is little bit hard to have some AT log file.
> 
> I use M10 as TCP Server.
> 
> What I can assume from that GSM library is that after start of the server object : 
>                AT+QILOCIP  (then parse received IP)
>                AT+QILPORT=”TCP”,81
>                AT+QISERVER
> 
> And after some time I want to change port number then I call function from object which will call:
>                (delay for 1 sec)
> +++        
> (delay for 1 sec)
> AT+QICLOSE
> And call another function for begin of the new object with different port number 82:
>                AT+QILOCIP  (then parse received IP)
>                AT+QILPORT=”TCP”,82
>                AT+QISERVER
> 
> Result is, that I still receive messages from port 81 and nothing from 82.
> 
> Currently I particularly solve my problem but it is not what I really want. 
> 
> 
> Regards,
> Peter Brezina

Here is answer from Quectel:

> So your command sequence should be like below;
> 
> What I can assume from that GSM library is that after start of the server object : 
>                AT+QILOCIP  (then parse received IP)
>                AT+QILPORT=”TCP”,81
>                AT+QISERVER
> 
> And after some time I want to change port number then I call function from object which will call:
>                (delay for 1 sec)
> +++        
> (delay for 1 sec)
> **AT+QISRVC=2
> **AT+QICLOSE
> ………………..                      //you should get “CLOSE OK”
> AT+QISSTAT                      //query the current Server Status
> ..................                       //you should get “ERROR” response
> 
> And call another function for begin of the new object with different port number 82:
>                AT+QILOCIP  (then parse received IP)
>                AT+QILPORT=”TCP”,82
>                AT+QISERVER
> AT+QISSTAT                       // you should get “S: LISTENNING” response
> AT+QILPORT?                   // you should get “TCP: 82” 
> 
> At this point client should reconnect to the Server which runs on the module by specifying the new port number as 82. 
> 
> Looking forward to your reply. Thanks !
> 
> Best Regards,
> Mehmet Cihangir  Field Application Engineer 

At the end - my suggestion how to do it in library (file "GSM3ShieldV1ClientProvider.cpp"):
```
//Disconnect TCP main function.
int GSM3ShieldV1ClientProvider::disconnectTCP(bool client1Server0, int id_socket)
{		
	// id Socket does not really mean anything, in this case we have
	// only one socket running
	theGSM3ShieldV1ModemCore.openCommand(this,DISCONNECTTCP);
	
	// If we are not closed, launch the command
//[ZZ]	if(theGSM3ShieldV1ModemCore.getStatus()==TRANSPARENT_CONNECTED)
//	{
		delay(1000);
		theGSM3ShieldV1ModemCore.print("+++");
		delay(1000);
		theGSM3ShieldV1ModemCore.genericCommand_rq(PSTR("AT+QISRVC=2"));//PBR modification
		theGSM3ShieldV1ModemCore.genericCommand_rq(PSTR("AT+QICLOSE"));
		theGSM3ShieldV1ModemCore.setStatus(GPRS_READY);
//	}
	// Looks like it runs everytime, so we simply flush to death and go on
	do
	{
		// Empty the local buffer, and tell the modem to XON
		// If meanwhile we receive a DISCONNECT we should detect it as URC.
		theGSM3ShieldV1ModemCore.theBuffer().flush();
		theGSM3ShieldV1ModemCore.gss.spaceAvailable();
		// Give some time for the buffer to refill
		delay(100);
		theGSM3ShieldV1ModemCore.closeCommand(1);
	}while(theGSM3ShieldV1ModemCore.theBuffer().storedBytes()>0);

	theGSM3ShieldV1ModemCore.unRegisterUMProvider(this);
	return theGSM3ShieldV1ModemCore.getCommandError();
}
```

brezinapeto, 2016-02-26:
What I did in my GSM libraries :
```

void GSM3MobileServerService::stop()
{
	
	// Review, should be the server?
	/*theGSM3MobileClientProvider->disconnectTCP(local1Remote0, mySocket);
	if(flags & GSM3MOBILESERVERSERVICE_SYNCH)
		waitForAnswer();
	theGSM3MobileClientProvider->releaseSocket(mySocket);*/
	theGSM3MobileServerProvider->disconnectTCPServer(mySocket); //PBR modification
	if(flags & GSM3MOBILESERVERSERVICE_SYNCH)
		waitForAnswer();
	theGSM3MobileClientProvider->releaseSocket(mySocket);
	mySocket = -1;
}
```

```
//Disconnect Server main function.   PBR modification
int GSM3ShieldV1ServerProvider::disconnectTCPServer(uint8_t socket)
{
	// id Socket does not really mean anything, in this case we have
	// only one socket running
	theGSM3ShieldV1ModemCore.openCommand(this,DISCONNECTTCP);
	
	// If we are not closed, launch the command
//[ZZ]	if(theGSM3ShieldV1ModemCore.getStatus()==TRANSPARENT_CONNECTED)
//	{
		delay(1000);
		theGSM3ShieldV1ModemCore.print("+++");
		delay(1000);
		theGSM3ShieldV1ModemCore.genericCommand_rq(PSTR("AT+QISRVC=2"));//PBR modification
		theGSM3ShieldV1ModemCore.genericCommand_rq(PSTR("AT+QICLOSE"));
		theGSM3ShieldV1ModemCore.setStatus(GPRS_READY);
//	}
	// Looks like it runs everytime, so we simply flush to death and go on
	do
	{
		// Empty the local buffer, and tell the modem to XON
		// If meanwhile we receive a DISCONNECT we should detect it as URC.
		theGSM3ShieldV1ModemCore.theBuffer().flush();
		theGSM3ShieldV1ModemCore.gss.spaceAvailable();
		// Give some time for the buffer to refill
		delay(100);
		theGSM3ShieldV1ModemCore.closeCommand(1);
	}while(theGSM3ShieldV1ModemCore.theBuffer().storedBytes()>0);

	theGSM3ShieldV1ModemCore.unRegisterUMProvider(this);
	return theGSM3ShieldV1ModemCore.getCommandError();
}
```

agdl, 2016-07-12:
This issue was moved to arduino-libraries/GSM#1

closed 2016-07-12 by agdl