arduino/Arduino #4617: #error being incorrectly processed in pre-processor

thomasonw, 2016-02-27:
What steps will reproduce the problem?
------------
Using the following test sketch:
`
#if defined(__AVR_AT90CAN32__) || \
    defined(__AVR_AT90CAN64__) || \
    defined(__AVR_AT90CAN128__)
        #define  CANMB_QUANTITY     15                     

#elif defined(__AVR_ATmega32C1__) || \
      defined(__AVR_ATmega64C1__) || \
      defined(__AVR_ATmega16M1__) || \
      defined(__AVR_ATmega32M1__) || \
      defined(__AVR_ATmega64M1__)
        #define  CANMB_QUANTITY     6                       

#else
     #error Unsupported CPU in avr_can.h
#endif

void setup() {
  Serial.begin(115200);
 
  Serial.println(CANMB_QUANTITY);
 
}

void loop() {

}

`

What is the expected output? What do you see instead?
------------
While compiling on an ATmega64M1 uC, the line "#error Unsupported CPU in avr_can.h" is reported by the compiler and compiling stops.  As there was a match I do not expect to #else lines to be recognized.   

What version of the product are you using? On what operating system?
------------
Arduino IDE  1.6.7
Windows 7

Please provide any additional information below.
------------

Additional testing, it seems the #if ...   statements are being processed correctly, in that the ATmega64M1 uC was recognized and CANMN_QUANITY was set to 6.   But then processing continues on and picked up the #error line.

And it seems it is only the #error that is incorrectly picked up.  If I place other directives (e.g., some test #defines) in the #else area, processing seems to work correctly.

per1234, 2016-02-27:
The issue is caused by https://github.com/arduino/arduino-builder/issues/106. I have submitted a pull request to fix this issue in your ATmegaCAN platform.txt https://github.com/thomasonw/ATmegaCAN/pull/1. The MCU specific macros in the example sketch above were converted to bold text by markdown. The correct code is:
```arduino

#if defined(__AVR_AT90CAN32__) || \
    defined(__AVR_AT90CAN64__) || \
    defined(__AVR_AT90CAN128__)
        #define  CANMB_QUANTITY     15                     

#elif defined(__AVR_ATmega32C1__) || \
      defined(__AVR_ATmega64C1__) || \
      defined(__AVR_ATmega16M1__) || \
      defined(__AVR_ATmega32M1__) || \
      defined(__AVR_ATmega64M1__)
        #define  CANMB_QUANTITY     6                       

#else
         #error Unsupported CPU in avr_can.h
#endif

void setup() {
  Serial.begin(115200);
 
  Serial.println(CANMB_QUANTITY);
 
}

void loop() {

}
```

thomasonw, 2016-02-27:
Thank you, confirmed the pull request solved this issue.

closed 2016-02-27 by thomasonw