arduino/Arduino #4619: Alternative avrdude.conf not reconized when using 'Upload using programmer' option.

thomasonw, 2016-02-28:
Problem
----------------
When selecting 'upload using programmer' targeting an alternative CPU, the command string passed to avrdude does not contain the path to the custom avrdud.conf file.   If using the USB port (normal way to upload sketches), the correct path is passed and all works well.  However, when using the programmer the 'standard' path is passed and  avrdude faults out with 'unrecognized CPU type' error.

Details
-----------------
If using USB port to upload a program, AVRdude is passed the following:
`C:\Program Files\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\Win7\AppData\Local\Arduino15\packages\ATmegaCAN\hardware\avr\1.0.2/avrdude.conf     -v -patmega64m1 -carduino -PCOM22 -b57600 -D -Uflash:w:C:\Users\Win7\AppData\Local\Temp\build2af98d057f4e31a77cc1e543f707ebe2.tmp/sketch_feb27a.ino.hex:i `

However, if 'upload using programmer' is selected, the following is passed:
`C:\Program Files\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega64m1 -cusbasp -Pusb -Uflash:w:C:\Users\Win7\AppData\Local\Temp\build2af98d057f4e31a77cc1e543f707ebe2.tmp/sketch_feb27a.ino.hex:i `

Environment
--------------------
Arduino IDE 1.6.7
Added in custom board script via board manager: 
  https://thomasonw.github.io/ATmegaCAN/package_thomasonw_ATmegaCAN_index.json
  (Support for ATmegaxxM1 uC's)
Win 7
Attached files:  platform.txt, boards.txt, and programmers.txt included in ATmegaCAN board download.

Additional 
---------------------
It appears to me that the relevant lines in platform.txt are:

* tools.avrdude.config.path={runtime.platform.path}/avrdude.conf
* tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"
* tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"

in that the last line:  tools.avrdude.program.pattern=...   is what is used when selecting 'Upload using programmer'?   And it also seems that the {config.path} is not being picked up correctly, using the default path vs. the one defined in the custom platform.txt file?   (I am only speculating here)

Related?
------------------
" board.*.tool in boards.txt and tools directives in platforms.txt are ignored #3463 "
[https://github.com/arduino/Arduino/issues/3463](url)

Attached
-------------

[programmers.txt](https://github.com/arduino/Arduino/files/149652/programmers.txt)
[platform.txt](https://github.com/arduino/Arduino/files/149653/platform.txt)
[boards.txt](https://github.com/arduino/Arduino/files/149654/boards.txt)

per1234, 2016-02-28:
Make sure you have the right programmer selected in **Tools > Programmer**. 

**Upload Using Programmer** with the second USBasp in **Tools > Programmer** gives me:
```
C:\Program Files (x86)\arduino-1.6.7\hardware\tools\avr/bin/avrdude -CC:\Users\per\AppData\Local\Arduino15\packages\ATmegaCAN\hardware\avr\1.0.2/avrdude.conf -v -patmega64m1 -cusbasp -Pusb -Uflash:w:C:\Users\per\AppData\Local\Temp\build2830f38d10547509fcabed146640609a.tmp/Blink.ino.hex:i 
```
I would recommend changing the name values in your programmers.txt to make it clear which of the **Tools > Programmer** entries are specific to your package(for example: `usbasp.name=USBasp(ATmegaCAN)`).

Adding copies of all the programmers is an effective workaround for the issue of platform.txt associated with the selected programmer being used for **Upload Using Programmer**(that's the solution that was used for #3463 in [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/programmers.txt)) but since these programmers are shown regardless of which board is selected it can really clutter up the **Tools > Programmer** menu if you have multiple packages installed that do this. It would be great if there was a better way to do this. One idea is to change the IDE so that if multiple programmers have the same ID in programmers.txt then the one associated with the selected board overrides the others.

I have a couple of custom programmer projects where I have the opposite issue of platform.txt associated with the selected board being used for **Tools > Burn Bootloader**. I'm not sure how it could be done but it would be really nice to have more control over this.

thomasonw, 2016-02-28:
Well, that sure seems awkward. . .   

 

I had not noticed the many repeats of programmers, but as you pointed out – selecting one further down and associated with the custom board port works as intended.   But sure does seem like an awkward way for Arduino IDE to handle these…   Perhaps rather than a bug report, this is more of a future enchantment suggestion?  Like your idea of not repeating programmers with the same .name ID, but making sure internally the correct scripting associated with the selected board is used?  Would reduce greatly the clutter in the programmers list.

 

Taking your suggestion and adding (ATmegaCAN) to the names, I tried it as you offered, and also like this:

usbasp.name=(ATmegaCAN)USBasp

 

FWIW, it seems to my eyes to read better putting the board name 1st, is there a string onion in the Arduino community for the ordering?

 

Again want to thank you for my continued education via problem reports – do appreciate your indulgence.

 

-al-

per1234, 2016-09-25:
The duplicate programmers workaround to this issue is really not ideal. If you have multiple 3rd party hardware packages that do this the **Tools > Programmer** menu gets really cluttered up and it's more confusing for the user to have to select the programmer associated with the board they're using.

Examples of 3rd party hardware packages that do this are:
- https://github.com/thomasonw/ATmegaxxM1-C1
- https://github.com/SpenceKonde/ATTinyCore
- https://github.com/MCUdude/MightyCore

I though of a better solution today: to override `tools.avrdude.config.path` in the 3rd party hardware package's boards.txt. Unfortunately the Arduino IDE does not currently allow this, though it's possible to override other properties in platform.txt by setting them in boards.txt. Ideally the platform.txt value would be used as the default but could be overridden if set elsewhere. This would allow 3rd party hardware packages to set the path to `{runtime.platform.path}/avrdude.conf`, as is currently done in all the packages I listed.

Another option would be for the location of avrdude.conf in Arduino AVR Boards to be moved to the avr folder and the platform.txt to be changed from:
```
tools.avrdude.config.path={path}/etc/avrdude.conf
```
to:
```
tools.avrdude.config.path={runtime.platform.path}/avrdude.conf
```

The root of this particular issue is that for some reason the avrdude.conf file is incomplete. You'd think that AVRDUDE would have added all AVR parts but this is not the case.

It seems inconsistent to me that there is one behavior for Burn Bootloader and another behavior for Upload Using Programmer. I've never seen the Arduino developers give a straight answer as to whether this is intentional or a bug even though this issue has been reported quite a few times here in various forms. It has certainly caused a lot of confusion and headaches. It was not documented until I added a note in the GitHub wiki about it.

per1234, 2020-12-24:
The issue of the redundant programmers has been rendered moot by https://github.com/arduino/Arduino/pull/9900

The issue of the behavior not being well documented has been resolved by https://github.com/arduino/arduino-cli/pull/925

The behavior of "Upload Using Programmer" and "Burn Bootloader" is consistent in Arduino CLI (see [the Arduino Platform Specification](https://arduino.github.io/arduino-cli/latest/platform-specification/) for details) and that improvement will eventually propagate to the other official development software (as well as some 3rd party software) over time.

So I'm going to close this as fixed. If there is an outstanding issue I have overlooked, feel free to comment here and I'll reopen.

closed 2020-12-24 by per1234