arduino/Arduino #4623: Arduino IDE 1.6.5 Mac OS X 10.11 -- Error downloading package_index.json

remct, 2016-02-28:
I cannot get the Board Manager under Arduino IDE 1.6.5 on a Mac OS X 10.11 system to load.
(My goal is to use that IDE to program a panstamp). 
I have installed the latest version of java:

java version "1.8.0_73"
Java(TM) SE Runtime Environment (build 1.8.0_73-b02)
Java HotSpot(TM) 64-Bit Server VM (build 25.73-b02, mixed mode)

I have disabled rootless SIP protection on my Mac as well. 
Can anyone help?

I get the following errors:

Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:158)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:100)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:279)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:255)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:237)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:154)
	... 1 more
Caused by: java.io.IOException: Recevied invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:160)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:98)
	... 5 more

sandeepmistry, 2016-03-10:
@remct could you please try with the [latest v1.6.8 release of the IDE](https://www.arduino.cc/en/Main/Software) or hourly.

remct, 2016-03-10:
Hi, thanks.  I needed to load 1.6.5 for my project. Later versions of the
IDE are documented as being problematic for my specific application
(panstamp).
I wound up abandoning the mac environment and am instead using Ubuntu.

sandeepmistry, 2016-03-10:
Ok, I'll close this for now, if someone encounters the same issue with the latest release we can re-open.

shanetoven, 2016-03-11:
Hi. Please re-open this issue. I'm experiencing the same.

sandeepmistry, 2016-03-11:
@shanetoven please provide the IDE version you are using and the stack trace.

shanetoven, 2016-03-11:
Arduino IDE 1.6.8, OS X 10.11.3.

 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more

shanetoven, 2016-03-11:
I tried the latest hourly build as well. Same result. Checked all proxy settings within the IDE and in my OS (I'm not using a proxy). No change. Entering the URL into a browser produces a valid return.

sandeepmistry, 2016-03-11:
Thank you for the information, re-opening.

shanetoven, 2016-03-11:
My workaround was to pull the desired board files manually via git, but clearly not optimal or the normal behavior for this feature.

sandeepmistry, 2016-03-14:
@shanetoven @remct

Could you please try deleting ``` ~/Library/Arduino15/package_index.json``` and restarting the IDE?
```
rm ~/Library/Arduino15/package_index.json*
```

Based on the following line ```Caused by: java.io.IOException: Received invalid http status code from server: 416``` the server is returning HTTP error code 416, which maps to "Requested Range not satisfiable". I'm wondering if the file exists on disk with a size that is larger than what's on the server.

jeff-pang, 2016-03-15:
Hi,

I am experiencing the same issue. removing package_index.json* did not help. I am using OSX El Capitan Version 10.11.3 with Arduino IDE 1.6.8

jeff-pang, 2016-03-15:
It's ok now. When I was experiencing the issue I tested the url (http://downloads.arduino.cc/packages/package_index.json), and the server wasn't responding, and then I tried again awhile later and the server is back, everything worked.

shanetoven, 2016-03-15:
I can confirm this worked for me as well.

 
  
 
 
  

  

https://github.com/arduino/Arduino/issues/4623#issuecomment-196813842

sandeepmistry, 2016-03-28:
@shanetoven did you have to use my suggestion from https://github.com/arduino/Arduino/issues/4623#issuecomment-196535250 or did it just start working?

shanetoven, 2016-03-28:
I used your suggestion. Deleting the files made it work.

--Shane

sandeepmistry, 2016-04-05:
@shanetoven thank you for confirming.

I'm going to close this for now, as we cannot reproduce it. However, if someone runs into the issue in the future please provide a copy of your ```~/Library/Arduino15/package_index.json*``` files. It might provide some clues as to what triggered this issue.

closed 2016-04-05 by sandeepmistry

apreble, 2016-07-20:
I'm having the same issue. IDE 1.6.9 MacOS 10.11.6. I've uploaded the file as suggested. Added txt to the end so it can be embedded here. Deleting it did not work.
[package_index.json.txt](https://github.com/arduino/Arduino/files/374342/package_index.json.txt)

` Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more
`

apreble, 2016-07-20:
Forgot to mention that I'm not using a proxy.

clivepengelly, 2016-07-30:
Yep. I've spent the best part of 2 days trying to get this to work. Please can someone shed some light on this problem. I am using Windows 10 - for my sins.

escapemyroom, 2016-07-30:
I eventually got it to work.  I had to delete the whole folder mentioned above and then it started working again.

clivepengelly, 2016-07-30:
[image]
I uninstalled and reinstalled Arduino, but same thing. I tried renaming the package_index.json, but it made no difference. This file is under Adruino\dist. I have now removed all Arduino folders under my documents folder and reinstalling the IDE. I don't think this will make any difference but I will keep trying. If you look at the screenshot, It loads a few boards under the board manager before crashing. I can reach the web site by placing that json url into the browser. Highly frustrating.

clivepengelly, 2016-07-30:
It looks like a security problem to me. If I try and install any of the listed boards it gives the same error. Unable to copy......
Where is it trying to copy these files to? I have tried setting the folder privileges to "everyone" full control. on all the Arduino folders where I can find them. Under program files and the user\local\appdata. Still no difference.

clivepengelly, 2016-07-30:
Some of these packages do install. Like Windows IOT. How do I get the ESP8266 into the list so that I can download it? http://arduino.esp8266.com/stable/package_esp8266com_index.json

clivepengelly, 2016-07-30:
Adding it to preferences does not work for some reason. I think the board manager crashes before it can get there.

GabrielRaya, 2016-08-19:
Do it manually!. Just copy the package_esp8266com_index.json file inside AppData\Local\Arduino15 Folder, then restart the IDE. Go to board manager and then you will see ESP8266 on the list, select it and install the latest version.

thegecko, 2016-08-20:
I had this problem when connected to a corporate VPN. Disconnecting and restarting/retrying fixed the issue.

hamid1234567, 2017-07-14:
I have the same issue .
Most probably this issue produced after java update to 8 . now i try all solutionin this page without any success
tried solutions:
1- delet temp of arduino15 folder
2- delete all arduino 15 folder (after many time unsuccessful other solution)
3- set java command by cmd to true

Please help me!!!

MissrRight, 2018-01-20:
I have the same issue.
But fortunately, I made it!
My experiences:
-I did delete the  `~/Library/Arduino15/package_index.json*`   as @sandeepmistry  suggested, but it didn't work for me.
-And I re-install the Arduino IDE, it didn't work.
-Finally, I delete all the files of Arduino15 ``~/Library/Arduino15`, and it just worked!!  :P
So, if you have the same issue, I don't know how you can exactly made it, but you can try the same ways.
Hope you make it. :)

qianfan-Zhao, 2018-06-30:
Hi:
I have the same problem on my arduino 1.8.1 on windown platform. But I can't find any directory named Arduino15. How to solve this problem? 

Thanks!
```
 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:334)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:309)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:287)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 5 more
```

qianfan-Zhao, 2018-07-17:
@facchinm Thanks and I found that is a hidden folder. 
It works on my computer.

Thanks!

amitvashist, 2019-01-26:
Deleting Arduino15 folder works for me.
In windows10 open powershell and write
rm -r C:\Users\yourUserName\AppDAta\Local\Arduino15.

Thanks

1162kpratik, 2021-04-14:
> I cannot get the Board Manager under Arduino IDE 1.6.5 on a Mac OS X 10.11 system to load.
> (My goal is to use that IDE to program a panstamp).
> I have installed the latest version of java:
>
> java [...]

- Enable VPN 
- Then restart your Arduino IDE
- Open board manager 
- It will start downloading :)