arduino/Arduino #4629: GSM Playground Shield Library

cniweb, 2016-02-29:
A library for the GSM Playground Shield:
http://www.hwkitchen.com/products/gsm-playground/

Library:
https://github.com/cniweb/gsm-playground

agdl, 2016-03-01:
The library can't be placed in the library manager because it is not compliant with other Arduino libraries. Please follow this guide
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cniweb, 2016-03-03:
What exactly is wrong?

cniweb, 2016-03-03:
Need I a src/ folder?

per1234, 2016-03-03:
The src folder is not necessary and if you want to retain backwards compatibility with Arduino IDE 1.0.x you shouldn't use one.

There are a couple of problems with your library:

Folder structure: you need to remove the GSM-Playground folder. A good way to test if your folder structure is correct is to [download the library](https://github.com/cniweb/gsm-playground/archive/master.zip) and then **Sketch > Include Library > Add .ZIP Library**. You will get a `Specified folder/zip file does not contain a valid library` error with the current folder structure.

Another issue is that there is already a GSM library included with the Arduino IDE. So if you try to compile ATcommandsTest.ino with a recent version of the Arduino IDE you will see the warning:
```
Multiple libraries were found for "GSM.h"
 Used: C:\Program Files (x86)\arduino-1.6.7\libraries\GSM
 Not used: E:\arduino\libraries\GSM-Playground
```
Because the folder name of your library is different from GSM.h and the folder name of the Arduino built-in GSM library is the same the built in library's GSM.h is included instead of the intended one which causes the compile to fail. If you rename your file it will solve this issue.

There's another minor issue relating to your folder structure you might consider, though it is not a required change: From https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification:
>When the user imports a library into their sketch (from the "Tools > Import Library" menu), an #include statement will be added for all header (.h) files in the src/ directory (but not its subfolders). As a result, these header files form something of a de facto interface to your library; in general, the only header files in the root src/ folder should be those that you want to expose to the user's sketch and plan to maintain compatibility with in future versions of the library. Place internal header files in a subfolder of the src/ folder.

With your current folder structure if you do **Sketch > Include Library > GSM-Playground** the Arduino IDE adds:
```arduino
#include <AT.h>
#include <GPS_GE863.h>
#include <GSM.h>
#include <GSM_GPRS.h>
#include <Setting.h>
```

cniweb, 2016-03-06:
@per1234 OK, I change my library, looks good now?

per1234, 2016-03-06:
Looks like the folder structure is now correct for Library Manager installation and the conflicting header filename has been changed. I submitted a pull request for some fixes to the examples. Another thing you might consider is to add a keywords.txt.

cniweb, 2016-03-10:
OK, I accepted your PR and add a Trvis-CI Build to verify the library under the Arduino Software.

agdl, 2016-03-11:
@cniweb please check your tag. Tag is 1.1.0 but properties  says 1.0.0

cniweb, 2016-03-13:
@agdl I update the Tag to 1.1.1

cniweb, 2016-03-29:
It´s now OK?

agdl, 2016-03-29:
Sorry. Done thank you!

closed 2016-03-29 by agdl