arduino/Arduino #4630: Rename modulo to remainder and add examples

cousteaulecommandant, 2016-03-01:
First, the Arduino reference refers to % as "modulo".  However "remainder" is a more appropriate (and familiar) name (and the one used in the C++ standard to describe what % does; also it's the third word that appears in the description in the Arduino documentation).
Second, the reference should include some examples of what happens when you % negative numbers.  Here's where the conceptual difference between "remainder" and "modulo" lies: the result of the former has the same sign as the FIRST operand, whereas the latter would have the one of the SECOND.  This is annoying since almost in 100% of the cases where % meets negatives you want the second behavior (e.g. `(-13) % 10` yielding `7`); however C++ surprises you with the first behavior (`-3`).  Therefore it is important to clarify this in the documentation.

TheAustrian, 2016-05-15:
I agree - especially since in the example it's promoted as a way to stay within the bounds of an array. Woe to those who try to count down instead of up in such an instance.

per1234, 2018-04-18:
Resolved by: https://github.com/arduino/reference-en/pull/350

closed 2018-04-27 by per1234