arduino/Arduino #4631: avrdude.conf does not differentiate between ATmega328 and ATmega328p, needs to support both

robatey, 2016-03-01:


facchinm, 2016-07-19:
will be solved by https://github.com/arduino/Arduino/pull/5021

darrylp1per, 2016-07-19:
Not solved by #5021 
the bootloader returns a values ( of SIGNATURE_0, 1 & 2,  that is what AVRdude from the IDE hears back from the chip.

So the bootloader needs to be altered to use either the atmega 328 or the atmega 328p values, which means altering 

.hardware/tools/avr/avr/include/avr/iom328p.h

with something like below 

/* Signature */
#define SIGNATURE_0 0x1E
#define SIGNATURE_1 0x95
#if defined(__AVR_ATmega328__)
#  define SIGNATURE_2 0x14
#else /* ATmega328P */
#  define SIGNATURE_2 0x0F
#endif

then fix the code of the bootloaders ( typically optiboot ) to alter its makefile, so it passes the right atmega_328 or atmega_328p chip id.  then it can be set.

ps. if altering iom328p.h

then consider altering these 

#if defined(__AVR_ATmega328P__)
#  define BODSE 5
#  define BODS 6
#endif
which only exist on the 328p pico chip.

then you have to have a boards.txt file that allows seletion of the two MCU types.

ie.  add this into boards.txt

menu.cpu=Processor

uno.menu.cpu.normal=AT Mega 328
uno.menu.cpu.normal.build.mcu=atmega328
uno.menu.cpu.pico=AT Mega 328P (Pico Power)
uno.menu.cpu.pico.build.mcu=atmega328p

facchinm, 2016-07-19:
Sorry, I misread 328p vs 328pb (which is the brand new topic).
However, I can't understand what the request is. Do you want to support another board or are the actual tools wrong? Because we never produced a board with plain 328. And of course if the optiboot output is wrong we should compile an optiboot for 328 (which we don't officially support)

facchinm, 2016-07-19:
Indeed, avr toolchain in PR #5021 contains
```
/* Signature */
#define SIGNATURE_0 0x1E
#define SIGNATURE_1 0x95
#if defined(__AVR_ATmega328__)
#  define SIGNATURE_2 0x14
#else /* ATmega328P */
#  define SIGNATURE_2 0x0F
#endif
```
Is it correct?

darrylp1per, 2016-07-19:
yes, the values are right.  for iom328p.h 

just the Makefile of optiboot to make either a 328  or 328p version. as needed.

that then stops the IDE complaining ( on running avrdude to upload a new sketch )  but the boards.txt file needs a menu option to select the actual atmel 328 or 328p varinat of the chip used.

many clones and people selling stand alone chips ( replacing blwn ones for example )  are selling the 328 not the 328p ( ala the pico low power variant that the genuine boards have. )

its only a prblem for someone buying a totally blank at mega 328   9 not P variant )  as they will struggle to burn a bootloader into it, as currently the AS  and the IDE only really know and compile programs for the 328P version of the chip.

I've manually hacked my files to allow me to program and upload by either ISP  or the IDE to both MCU versions.   But I'm not sure if its really needed for 99% of people in arduno land.

matthijskooijman, 2016-07-19:
I'm not sure if this is up to the Arduino IDE to fix - anyone that sells cloned Arduino boards is responsible for providing support, for example by providing a custom platform.txt and boards.txt for their customized boards. Adding a CPU selector to the official Arduino boards, which is never needed for official boards, will probably only cause confusion for users that *did* by official boards, so I'm not sure if this is something we should fix here. If there are changes needed in e.g. the java code that cannot be handled with a custom core, then that's something that would be worth looking into.

robatey, 2016-07-19:
Hi Martino,

so friendly .. thanks.  I actually develop stuff using "stand-alone" 28 pin ATMega chips.  The problem surfaced when I tried to install a bootloader on a 328 (instead of a 328p) processor.  I was doing this using a socketed UNO board.  I learned that 328's were cheaper than 328p's and executed the same code.  Once some of the tricks are learned, it is easy to use a breadboard with a chip instead of a standard (or clone) board.  The sketches work nicely for a lot of things, thus avoiding assembly language.

Robert Batey, Eagle Idaho

per1234, 2017-07-06:
Arduino AVR Boards supports the Arduino AVR Boards, none of which use ATmega328.

avrdude.conf contains definitions for both parts.

If you want to use the Arduino IDE with ATmega328 I recommend the excellent [MiniCore](https://github.com/MCUdude/MiniCore), which supports ATmega328 just fine, including bootloaders specifically compiled for ATmega328 so they return the correct signature.

closed 2017-07-06 by per1234