arduino/Arduino #4632: needs HEX mode of Serial Monitor

night-ghost, 2016-03-01:
Because not only text can be sent via serial port.

cousteaulecommandant, 2016-03-01:
This might be useful.  Some serial port terminals such as gtkterm have a similar feature.  The displayed format could be similar to the one from the `hd` command, displaying both hex and (extended) ASCII.

Avamander, 2016-03-13:
Maybe it could be possible to instead allow launching external terminal software? I don't think Arduino IDE's Serial Monitor is worth time spent on improving it. There is times better programs that could instead be launched by the IDE.

rin67630, 2019-03-16:
>There are times better programs that could instead be launched by the IDE.

...if that could be done user-friendly i.e. so that the serial port gets released to compile automagically.