arduino/Arduino #4634: Unable to launch Arduino IDE under mandatory profile in Win7

innovationDET, 2016-03-01:
So in our environment (schools) we use mandatory profiles for student accounts in Windows. Students are unable to launch the Arduino application on Win7 - the applications launches and then closes. Debug reveals: 
Loading configuration...
Settings issues: Arduino cannot run because it could no create a folder to store your settings.

Our IT guys say this is the down side of mandatory profiles because when modifying them - the shell folders often get rewritten on logoff as part of the tools used to manage them. In our case it is to the username of the account to manage them, so instead of being %appdata% & %userprofile%, it becomes hard coded to a directory that the students do not have access to (C:\users\xxxx). 

Looking at the code - it is evident that you prefer registry variables over the system environment variables, which is understandable. I have been advised this needs to be reversed when using mandatory profiles. 

A colleague suggested instead of rewriting the order for everyone - make a case to check if ntuser.dat is named ntuser.man (indicating a mandatory profile) and flip the order in that instance only.

facchinm, 2016-03-02:
Hi @innovationDET ,
probably the easiest solution is converting your IDE installations to portable format.
This can be accomplished by creating a folder called `portable` inside the IDE root, so the %userprofile% folders won't be used.
Can it possibly solve your issue?

innovationDET, 2016-03-02:
Thanks.

I believe it has. It came up initially stating Arduino cannot run because it could not create a folder to store your sketchbook, but that was due to permission inheritance on the folder.

closed 2016-03-02 by innovationDET