arduino/Arduino #4635: Documentation Error: sizeof -- see https://forum.arduino.cc/index.php?topic=383005.0

dpharris, 2016-03-01:
On https://www.arduino.cc/en/Reference/Sizeof the following example is wrong.  
```
for (i = 0; i < (sizeof(myInts)/sizeof(int)) - 1; i++) {
  // do something with myInts[i]
}
```
That < should be <= or the -1 should not be there, otherwise the last element in the array is never reached.

See: [(https://forum.arduino.cc/index.php?topic=383005.0)]

cousteaulecommandant, 2016-03-02:
Probably a copy-paste from the `for (i = 0; i < sizeof(myStr) - 1; i++)` above (which makes sense in that case because strings have a trailing \0 that should not be counted), or they somehow meant that the array was null-terminated too (which doesn't make much sense).

Also, I think the part where the null terminator is mentioned is in the wrong place.

agdl, 2016-03-10:
Done thank you! @cousteaulecommandant can you please open a new issue with the other suggestion?

closed 2016-03-10 by agdl