arduino/Arduino #4643: My request to add NeoPixelBus Library has yet to show up

Makuna, 2016-03-03:
I could not reopen the original https://github.com/arduino/Arduino/issues/4484, so I created a new issue. 

My library located at https://github.com/Makuna/NeoPixelBus has not yet shown up in the list after four days.  My other libraries showed up over night so I am concerned there is an issue.

sandeepmistry, 2016-03-03:
I can see it in the Library Manager on my Mac with IDE v1.6.7:

[image]

What IDE version and OS are you using?

Makuna, 2016-03-03:
v1.6.7 on Win10.  Your search above gives no results, I search for NeoPixel and below is what I get.

[image]

Search for Makuna, and this is all I get
[image]

sandeepmistry, 2016-03-03:
Ok, you're hitting this issue: https://github.com/arduino/Arduino/issues/4139#issuecomment-156857120

This issue has be resolved, however we have not released a new IDE version with the fix (https://github.com/arduino/Arduino/pull/4361).

Suggested workarounds for now:
 * Delete ```%LOCALAPPDATA%\Arduino15\library_index.json.tmp.gz``` and then restart the IDE.
 * Or try an hourly IDE build.

Makuna, 2016-03-03:
Ok, that works.

Is there an estimate as to when the next IDE will get published?  I have had a few users who raised this issue to me.

sandeepmistry, 2016-03-09:
IDE 1.6.8 is out now.

closed 2016-03-09 by sandeepmistry