arduino/Arduino #4654: virtualWire & Servo

jpl-pfungst, 2016-03-04:
#include <stdio.h>
#include <VirtualWire.h>   just by adding the include got the error below
#include <Servo.h>
----------------------------------------------------
C:\Program Files (x86)\Arduino\libraries\Servo\src\avr/Servo.cpp:81: multiple definition of `__vector_11'

libraries\VirtualWire\VirtualWire.cpp.o:C:\Program Files (x86)\Arduino\libraries\VirtualWire/VirtualWire.cpp:591: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren.
----------------------------------------------------------

Has some one solve the problem, I do not anderstand the code
thanks

PaulStoffregen, 2016-03-04:
On many AVR chips, these libraries use the same timer, so they're incompatible.

On some ARM chips, they use different timers and can be used together.

jpl-pfungst, 2016-03-05:
some poeple suggest to use the ServoTimer2 but this module does not compile!

Any suggestion are welcome

closed 2017-07-06 by per1234