arduino/Arduino #4660: The same code is works wrong when compiled with IDE v1.6.7 compared with 1.5.6 r2

radudorurus, 2016-03-06:
I wrote a program to read temperatures from DS18B20 sensors and display on a 20x4 LCD with a I2C backpack (PCF8574)
I have 5 screen modes.
The first one is displayed erratically.

[image]
[image]

I switched in the code
displayscreen01 with 02 and the problem still exist on the screen 02 which was 01 previously. 
Maybe the problem is from the WIRE library.

Here is the code for the screen:
`void displayscreen01 () // print 5 temperatures and TA state
{
  if (static_text_already_printed == 0) //print one time static text
  {
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print(F("Temperaturi"));
  lcd.setCursor(0,1);lcd.print(F("Tur:"));
  lcd.setCursor(0,2);lcd.print(F("Ret:"));
  lcd.setCursor(0,3);lcd.print(F("ACM:"));
  lcd.setCursor(10,1);lcd.print(F("Teu:"));
  lcd.setCursor(10,2);lcd.print(F("TA : "));
  lcd.setCursor(10,3);lcd.print(F("Ext:"));
  static_text_already_printed = 1;
  }
  //print dynamic items
  lcd.setCursor(12,0);
  timedisplay_hhmmss();
  // and temperatures follow
  lcd.setCursor(5,1); lcd.print(temp_turcazan,1); lcd.print(F(" "));
  lcd.setCursor(5,2); lcd.print(temp_returcazan,1); lcd.print(F(" "));
  lcd.setCursor(5,3); lcd.print(temp_bacm,1); lcd.print(F(" "));
  lcd.setCursor(15,1); lcd.print(temp_teu_tur_puffer,1); lcd.print(F(" "));
  lcd.setCursor(15,2);lcd.print(F("oooo")); 
  // full circle moving animation
  if (staretermostatambianta == LOW)
  {
    if (col_TA > 3)
    {
      col_TA = 0;
    }
    lcd.setCursor((15 + col_TA),2);
    lcd.write(byte(11)); // full circle
    col_TA += 1;
  }
  // lcd.setCursor(15,3); lcd.print(temp_ext,1); lcd.print(" ");
  char temp_ext_float[5]; dtostrf(temp_ext, 5, 1, temp_ext_float); // conversion float to string
  lcd.setCursor(14,3); lcd.print(temp_ext_float);
}

// print screen 02 with 5 temperatures
void displayscreen02 ()
{
  if (static_text_already_printed == 0) //print one time static text
  {
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print(F("Stare pufer"));
  lcd.setCursor(0,1);lcd.print(F("T"));lcd.print((char)13);lcd.print((char)13);lcd.print(F(":"));//char 13 is up arrow
  lcd.setCursor(0,3);lcd.print(F("T"));lcd.print((char)14);lcd.print((char)14);lcd.print(F(":"));//char 14 is down arrow
  lcd.setCursor(10,1);lcd.print(F("T"));lcd.print((char)126);lcd.print((char)9);lcd.print(F(":"));//char 9 is upright arrow
  lcd.setCursor(10,2);lcd.print(F("T"));lcd.print((char)126);lcd.print((char)126);lcd.print(F(":"));//char 126 is right arrow
  lcd.setCursor(10,3);lcd.print(F("T"));lcd.print((char)126);lcd.print((char)10);lcd.print(F(":"));//char 13 is downright arrow
  
  static_text_already_printed = 1;
  }
  //print dynamic items
  lcd.setCursor(12,0);
  timedisplay_hhmmss();
  // and temperatures follow
  lcd.setCursor(5,1);lcd.print(temp_puffersus,1); lcd.print(F(" "));
  lcd.setCursor(5,3);lcd.print(temp_pufferjos,1); lcd.print(F(" "));
  lcd.setCursor(15,1);lcd.print(temp_puffermijlocsus,1); lcd.print(F(" "));
  lcd.setCursor(15,2);lcd.print(temp_puffermijloc,1); lcd.print(F(" "));
  lcd.setCursor(15,3);lcd.print(temp_puffermijlocjos,1); lcd.print(F(" "));
}`

sandeepmistry, 2016-03-10:
@radudorurus if possible could you please share a full sketch, and also mention what library you are using for the LCD.

oqibidipo, 2016-03-11:
> `char temp_ext_float[5]; dtostrf(temp_ext, 5, 1, temp_ext_float); // conversion float to string`

That buffer is not large enough for a minimum of 5 characters plus a NUL terminator.

radudorurus, 2016-03-11:
@oqibidipo Yes it would be possible that the buffer is not long enough and the problem propagates from row 3 to row 0.
But how do you explain the error from row 2 columns 0-3? There is no buffer there. It is just printing clear text.

This is the library used for the LCD. It is v.1.2.1 from fmalpartida
[LiquidCrystal.zip](https://github.com/arduino/Arduino/files/168562/LiquidCrystal.zip)

This is the sketch, zip. I preferred zip because it is quite long (~1700 lines including description and comments)
[automatizare_cazan_new_latest.zip](https://github.com/arduino/Arduino/files/168565/automatizare_cazan_new_latest.zip)

oqibidipo, 2016-03-11:
> But how do you explain the error from row 2 columns 0-3? There is no buffer there. It is just printing clear text.

The memory layout of the LCD is a bit peculiar: row 2 follows immediately after row 0 and row 1 starts at address 64:
```
    0   1   2   3 ...  18  19
0   0   1   2   3 ...  18  19
1  64  65  66  67 ...  82  83
2  20  21  22  23 ...  38  39
3  84  85  86  87 ... 102 103
```

facchinm, 2016-07-19:
Duplicate of https://github.com/arduino/Arduino/issues/4302, solution [here](https://github.com/arduino/Arduino/issues/4302#issuecomment-164453037)

closed 2016-07-19 by facchinm