arduino/Arduino #4664: IDE Display problem on Ubuntu

psellers89, 2016-03-07:
I have a very strange display problem with the Arduino IDE on linux. The fonts become very strange. Maybe because I am using a hdpi screen (3200x1800). Check the screenshot attached. Thanks

http://s21.postimg.org/pfnyn1nsn/arduino_gui.png

lmihalkovic, 2016-03-07:
It might be because you seem to be working with a museum piece (1.0.5?)... time to embrace a more modern set of problems (there are many recent latest-build all with different things that work/dont work for different platforms - many being just minor annoyances).

psellers89, 2016-03-07:
Just installed 1.6.7. Everything is working fine now. Sorry I thought I was running the last release... Closed

closed 2016-03-07 by psellers89