arduino/Arduino #4670: Crash compiling after update to 1.6.8 - avr-g++: error: missing filename after '-o'

nisraelsen, 2016-03-10:
Previously working code now won't compile:

avr-g++: error: missing filename after '-o'

PaulStoffregen, 2016-03-10:
Do you really expect anyone can do anything useful without the actual code which produce the error?

PaulStoffregen, 2016-03-10:
If you do decide to share the code, also remember SimpleTimer.h is *not* one of the libraries installable from the library manager.  You must be specific about where to get this library, if you want anybody to be able to reproduce and investigate this issue.

denismoc, 2016-03-10:
I don't think the code is needed, no need to be so patronising.
I have the exact same problem and I tested several scripts of my own and even the provided examples. It must be a bug and hopefully the roll out an update to fix it soon.

facchinm, 2016-03-10:
Since we are not experiencing this bug, could you please send some debug infos (like verbose compilation output) and/or the failing sketch?

denismoc, 2016-03-10:
I just uninstalled and reinstalled it and it's working for me now.

PaulStoffregen, 2016-03-10:
Issues reports without details necessary to reproduce the problem wastes the valuable time of Arduino's developers and contributors.  That's limited time which could be much better used to further develop the software or fix other problems.

nisraelsen, 2016-03-10:
Here's my code.  Note that the SimpleTimer library is available at http://playground.arduino.cc/Code/SimpleTimer 

```
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>
#include <SimpleTimer.h>

// Data wire is plugged into port 6 on the Arduino
#define ONE_WIRE_BUS 6
#define FAN 8
#define Level_Pin 4
#define Pump_Pin 12
#define Inverter_Pin 2
#define Pump_Minutes 2
#define Off_Minutes 2

byte fantemp=75, n=0;
int fanTimer=0,inverterTimer, pumpOnTimer, inverterStopTimer, pumpWaitTimer;
boolean day=0, fanstate=0, wellFull=0;

SimpleTimer timer;

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

void stopFan ()
  {
    digitalWrite (FAN, HIGH);
    fanstate=false;
   }

void tempF (void)
{
  byte temp;
  
  
  sensors.requestTemperatures(); // Send the command to get temperatures
  //Serial.println("DONE");
  // We use the function ByIndex, and as an example get the temperature from the first sensor only.
  if (n==4)
  {
    Serial.print("Temperature: ");
    temp=(sensors.getTempFByIndex(0));
    Serial.print(temp);
    Serial.print ("  ");
  }
  ++fanTimer;
  if (temp>fantemp)
    {
      if (!fanstate)//fan is off
     {
      Serial.println ("Starting fan");
      digitalWrite (FAN, LOW);//turn it on
      fanstate=true;
     }
      fanTimer=0; //reset counter to keep fan on
      //deleted the fantimer Simpletimer because it kept other functions like pump from running their timers
    //if(timer.isEnabled(fanTimer)) //if fan is already on, then restart the timer to leave it on
    //  timer.restartTimer(fanTimer);   
    }
     

   else if ((fanTimer>11))
      {
        fanTimer=1; 
        if (fanstate)  //if fan is on
          {
            Serial.println();
          Serial.println ("Stopping fan");
          stopFan();
          }
      }
}

void startPump()
  {
   digitalWrite (Pump_Pin, LOW);
   Serial.println ("Starting pump");
   pumpOnTimer = timer.setTimeout((Pump_Minutes*6000), stopPump); //!!!!!!!!change to 60,000 when done debugging
  }

void stopPump ()
  {
   digitalWrite (Pump_Pin, HIGH);
  Serial.println ("Stopping pump");
  inverterStopTimer = timer.setTimeout(3000, stopInverter);//Change to 20000; stop inverter 20 seconds after pump stops
  pumpWaitTimer = timer.setTimeout(Off_Minutes*6000, tankLevel); //!!!!!!!!!!change to 60,000 when done debugging
  Serial.println ("Pump Wait Timer started");
  }

void tankLevel ()  //checks level of tank
  {
   if (!(digitalRead(Level_Pin))) //if pin is low, fill tank
    {
      if((!timer.isEnabled(pumpOnTimer)&&(!timer.isEnabled (inverterTimer))&&(!timer.isEnabled(pumpWaitTimer))&&(day)&&(wellFull)))
      {
        inverterTimer = timer.setTimeout(3000, startPump); //!!!!!change to 20000 after debugging
        digitalWrite (Inverter_Pin, HIGH); //turn on inverter      
        Serial.println ("Starting inverter");
      }
    }
  }

void stopInverter ()
{
  digitalWrite (Inverter_Pin, LOW);
  Serial.println ("Stopping inverter");
}

void daytime ()
  {
    int light=(analogRead (A4));
    if (light<300)
       {
        day=true;
        if (n==4)
        {
          Serial.print (" Day ");
          Serial.println (light);
        }
      }
      
    else
      {
       
        day=false;
        if (n==4)
        {
        Serial.print (" Night ");
        Serial.println (light);
        }
      }   
  }
  
void wellLevel ()
  {
    int level=(analogRead (A5));
    if (level<500)
      {
        wellFull=true;
        if (n==4)
        {
          Serial.print ("Well is Full ");
          Serial.println (level);
        }
      }
    else
      {
        wellFull=false;
        if (n==4)
        {
          Serial.print ("Well is Empty ");
          Serial.println (level);
        }
      }   
  }

void setup(void)
{
  // start serial port
  Serial.begin(9600);
  Serial.println("Well Level Controller Demo");
  
  //setup pins
  pinMode (FAN, OUTPUT);
  pinMode (Level_Pin, INPUT_PULLUP);
  pinMode (Inverter_Pin, OUTPUT);
  pinMode (Pump_Pin,OUTPUT);
  //digitalWrite (Level_Pin, HIGH);
  digitalWrite (FAN, HIGH);
  digitalWrite (Pump_Pin, HIGH);
  digitalWrite (Inverter_Pin, LOW);
  
  // Start up the library
  sensors.begin();
  
}

/*
 * Main function, get and show the temperature
 */
void loop(void)
{ 
  timer.run();
  tankLevel();
  wellLevel();
  tempF();
  daytime();
  ++n;
  if (n>4)
    n=0;
  delay (1000);
}
```

nisraelsen, 2016-03-10:
Here's the verbose compiler output:

```
Arduino: 1.6.6 (Windows 10), Board: "Arduino Nano, ATmega328"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\NAI\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\NAI\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "F:\Documents\Arduino\libraries" -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10606 -build-path "C:\Users\NAI\AppData\Local\Temp\buildee82f77835f99e794bbc4c186dab3fb3.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "F:\Documents\Arduino\Well controller\Well_control\Well_control.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\NAI\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\NAI\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "F:\Documents\Arduino\libraries" -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10606 -build-path "C:\Users\NAI\AppData\Local\Temp\buildee82f77835f99e794bbc4c186dab3fb3.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "F:\Documents\Arduino\Well controller\Well_control\Well_control.ino"
WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\eightanaloginputs" "C:\Users\NAI\AppData\Local\Temp\buildee82f77835f99e794bbc4c186dab3fb3.tmp\sketch\Well_control.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\eightanaloginputs" "-IF:\Documents\Arduino\libraries\OneWire" "-IF:\Documents\Arduino\libraries\Arduino-Temperature-Control-Library-master" "-IF:\Documents\Arduino\libraries\SimpleTimer-master" "C:\Users\NAI\AppData\Local\Temp\buildee82f77835f99e794bbc4c186dab3fb3.tmp\sketch\Well_control.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\eightanaloginputs" "-IF:\Documents\Arduino\libraries\OneWire" "-IF:\Documents\Arduino\libraries\Arduino-Temperature-Control-Library-master" "-IF:\Documents\Arduino\libraries\SimpleTimer-master" "C:\Users\NAI\AppData\Local\Temp\buildee82f77835f99e794bbc4c186dab3fb3.tmp\sketch\Well_control.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\eightanaloginputs" "-IF:\Documents\Arduino\libraries\OneWire" "-IF:\Documents\Arduino\libraries\Arduino-Temperature-Control-Library-master" "-IF:\Documents\Arduino\libraries\SimpleTimer-master" "C:\Users\NAI\AppData\Local\Temp\buildee82f77835f99e794bbc4c186dab3fb3.tmp\sketch\Well_control.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\NAI\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\eightanaloginputs" "-IF:\Documents\Arduino\libraries\OneWire" "-IF:\Documents\Arduino\libraries\Arduino-Temperature-Control-Library-master" "-IF:\Documents\Arduino\libraries\SimpleTimer-master" "F:\Documents\Arduino\libraries\OneWire\OneWire.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\NAI\AppDat [... cut ...] 82f77835f99e794bbc4c186dab3fb3.tmp\sketch\Well_control.ino.cpp" -o ""
avr-g++: error: missing filename after '-o'

Using library OneWire at version 2.3 in folder: F:\Documents\Arduino\libraries\OneWire 
Using library Arduino-Temperature-Control-Library-master at version 3.7.5 in folder: F:\Documents\Arduino\libraries\Arduino-Temperature-Control-Library-master 
Using library SimpleTimer-master in folder: F:\Documents\Arduino\libraries\SimpleTimer-master (legacy)
exit status 1
Error compiling.
```

facchinm, 2016-03-10:
Ok, the problem is that you are using IDE 1.6.6 with AVR core 1.6.10.
Updating to IDE 1.6.8 will surely fix the issue, or alternatively downgrading AVR core to 1.6.8 (via Board Manager) will have the same effect

nisraelsen, 2016-03-10:
Updating the IDE fixed the problem.  Thanks much!

closed 2016-03-10 by facchinm

KritanDotel, 2023-07-11:
Facing the same problem any alternative solutions to fix this problem??

per1234, 2023-07-11:
@KritanDotel this is not the appropriate place to request assistance. Please post a detailed description of your problem on Arduino Forum:

https://forum.arduino.cc/

I'm sure we'll be able to help you out over there.