arduino/Arduino #4672: slow compilation with Arduino IDE 1.6.7 on windows

vlast3k, 2016-03-10:
with arduino 1.6.5, compiling a pre-built sketch was almost instantaneous.  Now with 1.6.7 each time all files are recompiled and the previous results are not reused. Or maybe not all but it takes much more time

Attaching a verbose output of the compilation which was triggred exactly after a previous compile

```

C:\develop\arduino-1.6.7\arduino-builder -dump-prefs -logger=machine -hardware "C:\develop\arduino-1.6.7\hardware" -hardware "C:\Users\i024148\AppData\Local\Arduino15\packages" -tools "C:\develop\arduino-1.6.7\tools-builder" -tools "C:\develop\arduino-1.6.7\hardware\tools\avr" -tools "C:\Users\i024148\AppData\Local\Arduino15\packages" -built-in-libraries "C:\develop\arduino-1.6.7\libraries" -libraries "c:\develop\arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10607 -build-path "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "c:\develop\arduino\simple-co2-monitor-fw\simple-co2-monitor-fw.ino"
C:\develop\arduino-1.6.7\arduino-builder -compile -logger=machine -hardware "C:\develop\arduino-1.6.7\hardware" -hardware "C:\Users\i024148\AppData\Local\Arduino15\packages" -tools "C:\develop\arduino-1.6.7\tools-builder" -tools "C:\develop\arduino-1.6.7\hardware\tools\avr" -tools "C:\Users\i024148\AppData\Local\Arduino15\packages" -built-in-libraries "C:\develop\arduino-1.6.7\libraries" -libraries "c:\develop\arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10607 -build-path "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "c:\develop\arduino\simple-co2-monitor-fw\simple-co2-monitor-fw.ino"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp\sketch\simple-co2-monitor-fw.ino.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\RunningAverage" "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp\sketch\simple-co2-monitor-fw.ino.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\RunningAverage" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial" "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp\sketch\simple-co2-monitor-fw.ino.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\RunningAverage" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial" "-Ic:\develop\arduino\libraries\Streaming" "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp\sketch\simple-co2-monitor-fw.ino.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\RunningAverage" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial" "-Ic:\develop\arduino\libraries\Streaming" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\EEPROM" "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp\sketch\simple-co2-monitor-fw.ino.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\Streaming" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\EEPROM" "-Ic:\develop\arduino\libraries\Adafruit_NeoPixel" "-Ic:\develop\arduino\libraries\RunningAverage" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial" "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp\sketch\simple-co2-monitor-fw.ino.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\RunningAverage" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial" "-Ic:\develop\arduino\libraries\Streaming" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\EEPROM" "-Ic:\develop\arduino\libraries\Adafruit_NeoPixel" "-Ic:\develop\arduino\libraries\U8glib" "C:\Users\i024148\AppData\Local\Temp\build695f5752151071ccbfeb8ef910558f54.tmp\sketch\simple-co2-monitor-fw.ino.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\Adafruit_NeoPixel" "-Ic:\develop\arduino\libraries\U8glib" "-Ic:\develop\arduino\libraries\RunningAverage" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial" "-Ic:\develop\arduino\libraries\Streaming" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\EEPROM" "c:\develop\arduino\libraries\RunningAverage\RunningAverage.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\cores\arduino" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\variants\standard" "-Ic:\develop\arduino\libraries\Streaming" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\EEPROM" "-Ic:\develop\arduino\libraries\Adafruit_NeoPixel" "-Ic:\develop\arduino\libraries\U8glib" "-Ic:\develop\arduino\libraries\RunningAverage" "-IC:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial" "C:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\SoftwareSerial\SoftwareSerial.cpp" -o "nul"
"C:\develop\arduino-1.6.7\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\develop\arduino-1.6.7\hardware\ [... cut ...]
Using library Streaming in folder: c:\develop\arduino\libraries\Streaming (legacy)
Using library EEPROM at version 2.0 in folder: C:\develop\arduino-1.6.7\hardware\arduino\avr\libraries\EEPROM 
Using library Adafruit_NeoPixel in folder: c:\develop\arduino\libraries\Adafruit_NeoPixel (legacy)
Using library U8glib in folder: c:\develop\arduino\libraries\U8glib (legacy)

Sketch uses 25,054 bytes (77%) of program storage space. Maximum is 32,256 bytes.
Global variables use 971 bytes (47%) of dynamic memory, leaving 1,077 bytes for local variables. Maximum is 2,048 bytes.
```

PaulStoffregen, 2016-03-10:
Those many compiles are the library dependency detection.  It's not compiling code, only analyzing it to discover which libraries you are using.

Unfortunately, the builder is not yet smart enough to detect when it has discovered all the info, so it needlessly repeats the detection process.  Hopefully future versions will become smarter about minimizing the number of steps, and maybe even cache the results?

vlast3k, 2016-03-10:
ahhh i understand, so this is prehaps due to the modified compilation after 1.6.5 ?
indeed - it is faster than when compiling the first time, but on a bigger sketch (especially when compiling for ESP8266, where tehre is more space) it takes almost 30 minutes to discover that i have forgotten some semicolon :))

matthijskooijman, 2016-03-10:
@PaulStoffregen Do you mean it needlessly repeats detection within one compilation? Or only between compilation? Adding some caching between compilations seems like a good idea indeed - it can probably use the existing .d files to invalidate the cached values when a file changes.

closed 2016-03-13 by vlast3k

Avamander, 2016-03-13:
Why was this closed? It's worth doing if it makes the IDE faster and less resource heavy.

vlast3k, 2016-03-13:
i have closed it as i noticed that for the ESP8266 on both 1.6.5 and 1.6.7 the linker behaves the same way. For AVR projects on 1.6.5 the results seemed to be reused better indeed, but my pain point is the esp8266 as the sketches are much bigger there.

matthijskooijman, 2016-03-14:
@Avamander Agreed, but since this is something arduino-builder should fix, I've created a new issue here: https://github.com/arduino/arduino-builder/issues/127