arduino/Arduino #4678: Arduino 1.6.8 cant compile or download anything

iweryn, 2016-03-11:
hi, i wrote a program for my company but when i want to compile it and upload it to my arduino Due it says : exec: "/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Arduino Due (Programming Port).

and when i try to download anything either from board manager or libraries manager i get this error lines ` Chyba stahování http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Chyba stahování http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Chyba stahování http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:147)
	... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: java.security.cert.CertificateException: No subject alternative names present
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1949)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:302)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:296)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1509)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:216)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:979)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:914)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1062)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1375)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1403)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1387)
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:153)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:189)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more
Caused by: java.security.cert.CertificateException: No subject alternative names present
	at sun.security.util.HostnameChecker.matchIP(HostnameChecker.java:144)
	at sun.security.util.HostnameChecker.match(HostnameChecker.java:93)
	at sun.security.ssl.X509TrustManagerImpl.checkIdentity(X509TrustManagerImpl.java:455)
	at sun.security.ssl.X509TrustManagerImpl.checkIdentity(X509TrustManagerImpl.java:436)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:200)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:124)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1491)
	... 18 more
`

facchinm, 2016-07-19:
Hi @iweryn,
which version of the IDE are you using? It looks like there are SSL errors in the json request so it doesn't complete correctly. Check for Java conflicts (if you are on Linux) and/or proxy settings. About the first error, you need the toolchain to compile for Due, it's quite strange that you can select the board from the menu but the toolchian is not there. 
Anyway, check the latest IDE and please report here the errors you may find, thanks!

sandeepmistry, 2016-08-24:
I'm closing this due to lack of activity, please re-open with requested info. if you are still interested.

closed 2016-08-24 by sandeepmistry