arduino/Arduino #4682: Arduino due SAM Boards - Pin 13 malfunction

formicamania, 2016-03-13:
Hi,

ia am using the Arduino Due with SAM core 1.6.6, after the upgrade to 1.6.7 the Pin 13 in working as previously.

With analogwrite(13, xxx) the value is everytime high.

regards

sandeepmistry, 2016-03-14:
@formicamania please share an example sketch to show the behaviour you are seeing.

I am not seeing the issue with the following sketch, the built-in LED pulses.

```arduino
const int pin = 13;

void setup() {
  pinMode(pin, OUTPUT);
}

void loop() {
  for (int i = 0; i < 256; i++ ) {
    analogWrite(pin, i);

    delay(20);
  }

  for (int i = 256; i >= 0; i-- ) {
    analogWrite(pin, i);

    delay(20);
  }
}
```

However, there were some changes introduced in this area in #3524.

sandeepmistry, 2016-07-18:
Closing this due to lack of feedback.

closed 2016-07-18 by sandeepmistry